polhenarejos/pico-fido #3: credential management fails

al-heisner, 2022-11-25:
When using fido2-token to list / manage credentials stored, I'm receiving an error from fido2-token. Same command does work with a retail token.

# fido2-token -L -r /dev/hidraw0
Enter PIN for /dev/hidraw0: 
fido2-token: fido_credman_get_dev_rp: FIDO_ERR_RX

other info:
# lsusb | grep HSM
Bus 008 Device 108: ID cafe:4231 Pol Henarejos Pico HSM HID

# fido2-token -L 
/dev/hidraw0: vendor=0xcafe, product=0x4231 (Pol Henarejos Pico HSM HID)

al-heisner, 2022-11-26:
Looks like method CTAP_CREDENTIAL_MGMT (0x0A) is implemented in the token, but fido2-token is attempting to use authenticatorCredentialManagement (0x41) which is not implemented in the token.

fido2-token (libfido2) versions: 1.10.0,  1.12.0

polhenarejos, 2022-11-28:
This a preview feature, superseded in CTAP2.1. Note that the platform should return "not available" or similar, as it is not broadcasted in getInfo.

BTW, fixed in ae42e283

closed 2022-11-28 by polhenarejos