polhenarejos/pico-fido #4: waveshare rp2040 zero: LED do not blink

bobwxc, 2023-02-01:
Hi,

With the newest uf2 file, its LED didn't show anything.

GP 16 <-> DIN WS2812 RGB LED

[waveshare-rp2040-zero pin](https://www.waveshare.net/w/upload/c/c7/RP2040-Zero-2.jpg)

And testing on [demo.yubikey.com](https://demo.yubico.com/webauthn-technical/registration) is also no reaction.

```
[15096.929257] usb 1-4: new full-speed USB device number 48 using xhci_hcd
[15097.081523] usb 1-4: New USB device found, idVendor=feff, idProduct=fcfd, bcdDevice= 3.04
[15097.081545] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[15097.081552] usb 1-4: Product: Pico HSM HID
[15097.081557] usb 1-4: Manufacturer: Pol Henarejos
[15097.081563] usb 1-4: SerialNumber: ExxxA4931xxxxxxx
[15097.092666] hid-generic 0003:FEFF:FCFD.0015: hiddev1,hidraw4: USB HID v1.11 Device [Pol Henarejos Pico HSM HID] on usb-0000:00:14.0-4/input0
```

polhenarejos, 2023-02-05:
Maybe related with https://github.com/polhenarejos/pico-hsm/issues/3

Waveshare boards have a delayed ROSC initialization.

Try to compile as https://github.com/polhenarejos/pico-hsm/issues/4

```
export PICO_SDK_PATH=~/Devel/pico/pico-sdk
export USB_VID=0x234b
export USB_PID=0x0000
export PICO_BOARD=waveshare_rp2040_zero 
cmake .. -DENABLE_DELAYED_BOOT=1 -DPICO_BOARD=$PICO_BOARD -DUSB_VID=${USB_VID} -DUSB_PID=${USB_PID}
make -j4
```

bobwxc, 2023-02-08:
patched pico-sdk as https://github.com/polhenarejos/pico-hsm/issues/3

``` diff
diff --git a/src/boards/include/boards/waveshare_rp2040_zero.h b/src/boards/include/boards/waveshare_rp2040_zero.h
index 281b4dc..d1758ce 100644
--- a/src/boards/include/boards/waveshare_rp2040_zero.h
+++ b/src/boards/include/boards/waveshare_rp2040_zero.h
@@ -16,6 +16,11 @@
 // For board detection
 #define WAVESHARE_RP2040_ZERO
 
+// On some samples, the xosc can take longer to stabilize than is usual
+#ifndef PICO_XOSC_STARTUP_DELAY_MULTIPLIER
+#define PICO_XOSC_STARTUP_DELAY_MULTIPLIER 64
+#endif
+
 // --- UART ---
 #ifndef PICO_DEFAULT_UART
 #define PICO_DEFAULT_UART 0
```

and compiled with
``` shell
PICO_SDK_PATH=../pico-sdk cmake .. -DENABLE_DELAYED_BOOT=1 -DPICO_BOARD=waveshare_rp2040_zero -DUSB_VID=0x234b -DUSB_PID=0x0000
make
```

still no blink and no reaction.

This board is really confusing.

``` dmesg
[ 2044.086964] usb 1-4: new full-speed USB device number 11 using xhci_hcd
[ 2044.237292] usb 1-4: New USB device found, idVendor=234b, idProduct=0000, bcdDevice= 3.04
[ 2044.237303] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2044.237309] usb 1-4: Product: Pico HSM HID
[ 2044.237313] usb 1-4: Manufacturer: Pol Henarejos
[ 2044.237318] usb 1-4: SerialNumber: Exxxxxxxxxxxxxxxxx
[ 2044.241429] hid-generic 0003:234B:0000.0007: hiddev1,hidraw4: USB HID v1.11 Device [Pol Henarejos Pico HSM HID] on usb-0000:00:14.0-4/input0
```

polhenarejos, 2023-02-11:
I am using a tiny2040 and I cannot reproduce it. What I can recommend you:

- Ensure that your pico is completely blank. Do a pico_nuke to ensure it.
- `git pull` your `pico-fido` repo AND `git pull` of `pico-hsm-sdk` in it.
- Compile with delay option enabled.
- Since you are using the latest version, it should respond to `opensc-tool -an` with a message like this:
```
Using reader with a card: Yubico YubiKey OTP+FIDO+CCID
3b:fd:13:00:00:81:31:fe:15:80:73:c0:21:c0:57:59:75:62:69:4b:65:79:40
Unsupported card
```

bobwxc, 2023-02-11:
I'll try again.Thank you for your kind help!

closed 2023-02-21 by bobwxc