polhenarejos/pico-fido #5: Help Needed setting up a Raspberry Pi Pico RP2040

ryanamaral, 2023-02-17:
Thanks for making this library open-source 🚀 

I would like to start using it on my Raspberry Pi Pico RP2040 but I need some help identifying what I'm doing wrong.

Steps I've taken:
1. I started by flashing my pico with the existing build `pico_fido_pico-2.8.uf2`
2. The led starts blinking
3. When I then start the registration process the LED goes off 
4. When I click the button nothing happens

I've tried at https://www.token2.com/tools/fido2-test/ and https://demo.yubico.com/webauthn-technical/registration

Screenshots
[image]
[image]
[image]

polhenarejos, 2023-02-17:
If the LED goes off permanently it means it crashed for some reason. 

Which Pico model you use? Note the file you downloaded is just for the Pico model manufactured by the Raspberry Foundation https://www.raspberrypi.com/products/raspberry-pi-pico/
Have you nuked first? It is necessary to blank flash memory first. https://github.com/polhenarejos/pico-nuke
Can you try with the newest release 2.10? It fixes lots of bugs and could be related with yours.
I usually use https://webauthn.io/ for testing.

ryanamaral, 2023-02-18:
> Which Pico model you use? Note the file you downloaded is just for the Pico model manufactured by the Raspberry Foundation

That's exactly the one I'm using here.

> Have you nuked first? It is necessary to blank flash memory first.

Not really, didn't know that was a step until you made me aware.

> Can you try with the newest release 2.10? It fixes lots of bugs and could be related with yours.

Sure, I've followed the steps you've provided, nuked it first, and then flashed the latest version 2.10, and it is working! 

>  I usually use https://webauthn.io/ for testing.

Looks good, thanks. I'm using that one from now on.

Thanks for your help 🙇‍♂️

closed 2023-02-18 by ryanamaral