polhenarejos/pico-fido #7: Question about upgrade to a new version

aleytus, 2023-03-18:
Is it possible to upgrade to a new version without losing the binding, or is it necessary to unbind the device each time?

aleytus, 2023-03-19:
Also wondering why "mbedtls_x509write_crt_set_validity" is only up to "20320831235959", since fido (u2f) has no time limit!
https://github.com/solokeys/solo1/blob/master/tools/gencert/genca.sh

polhenarejos, 2023-03-20:
Depending the version you are using. The seed is never modified (unless you wipe your device) and residential keys are stored at the end of flash memory. However, the procedures of generating keys have been modified from time to time (specially first versions), which could made incompatible key loading of older keys. If you started with recent version you could upgrade by just dragging the new uf2 file.

I've increased validity up to 50 years in 59ec9b75.

closed 2023-03-20 by aleytus