polhenarejos/pico-fido #8: I lost access to my account in 2 acts

aleytus, 2023-05-23:
I use 2 factor authorization
[image]

Google recently added support for WebAuthn
So I decided to connect it
Then I deleted the key and rebinded it to add support for WebAuthn
[image]

And after that. I can't get into my account...
[image]

The 2-factor authorization also does not work
[image]

aleytus, 2023-05-23:
I found a bug on google...
If you register key at the beginning in FIDO2 and then the same key in U2F, then WebAuthn and 2FA work, but only through U2F mode
[image]
[image]

<summary>Spoiler!</summary>

aleytus, 2023-05-23:
PS. Access to the account was restored...

polhenarejos, 2023-05-24:
Glad to see you recover it.
Does the login through U2F work at least?

aleytus, 2023-05-24:
I want to clarify... Were you able to reproduce the error?
The thing is that I changed the code of v2.10 firmware to suit my needs.
After I encountered the error I flashed the original latest firmware, but maybe some changes in the spi flash might have remained...

aleytus, 2023-05-24:
I am also wondering how you debug the fido protocol. (how you intercept data for diagnostics)
I want to try to figure out the problem...

aleytus, 2023-10-12:
"WebAuthn" Works!
The key must be created through "Security keys"

[image]

closed 2023-10-12 by aleytus