polhenarejos/pico-fido #10: [Feature request] Support for NFC modules

dimaguy, 2023-07-14:
Would it be feasible to add NFC module compatibility to this program? So it can easily communicate with smartphones (ignoring eventual power supply issues here)

polhenarejos, 2023-08-14:
Pico W brings support for Wifi and BLE, but not NFC. NFC supports must be supplied by external modules, which will increase the complexity of the overall key.

PN532 seems the most popular solution, but I am not sure on writing capabilities. It seems that only works as a reader. Furthermore, from my experience, working with closed terminals is a nightmare, specially with iPhones. Apple usually adds fuzzy layers in between that require some not-well-documented tricks. 

In summary, to add support NFC it would require an NFC module tied to a computer and a non-negligible effort to support multiple phones, which deviates the original focus of Pico Fido on having a personal key.

dimaguy, 2023-08-14:
Yeah, you're right, NFC is too much of a mixed bag to be done. I'll just close this for now. Thanks for taking your time to reply

closed 2023-08-14 by dimaguy