polhenarejos/pico-fido #12: issue as yubikey5

alessiol78, 2023-07-20:
I try release 5.4 as yubikey5 (fw patched with portal https://www.picokeys.com/pico-patcher/): pico_fido_pico-5.4.yubikey5.uf2
I have Win10 (10.0.19045 Build 19045) and I'm using YubiKey Manager 1.2.5. The FIDO2 work as passkey but when I try to use manager it see the key but the items showed in Applications menu are grey and Interfaces menu not work.
The app show a message: "Make sure the application has the required permissions" (I runned it as Administrator). In user mode the app not see the fidokey.
[image]

polhenarejos, 2023-08-14:
It seems you are not using latest firmware  5.4, as it shows 3.0.0. It should appear as
[image]

alessiol78, 2023-08-15:
UPDATE: Using the key with Linux the firmware version is 5.4.0 but Yubico FIDO2 don't work. The OTP function work good. The same key in Windows report version 3.0.0 the OTP function don't work else FIDO2 work.

polhenarejos, 2023-08-16:
`main` branch contains frozen achievements. 
Try using `development` branch and specify only 1 USB_VID and USB_PID (the ones that Yubikey 5 uses).

polhenarejos, 2023-10-17:
Which version do you use? I made lots of changes in version 5.6

641i130, 2023-10-17:
Most recent build. Maybe Yubikey has caught on to people trying to spoof their products?

polhenarejos, 2023-10-20:
Firmwares should work on any RPi Pico. If it is detected by your host, it seems it is working. 
My point is that I do not understand why version 3.0.0 is displayed, even if you are using firmware 5.6, which has hard-coded the version and 3.0.0 does not appear at any place. Actually, when I run Yubikey app it displays 5.6 and I can operate with it normally.

Try to use ykman and report here what `opensc-tool -an` shows.

641i130, 2023-10-21:
On Windows 11 this is what happens for me:
[image]

```bash
# opensc-tool -an
Using reader with a card: Yubico YubiKey OTP+FIDO+CCID [Pico Key CCID Interface] (4150323833373205) 00 00
3b:fd:13:00:00:81:31:fe:15:80:73:c0:21:c0:57:59:75:62:69:4b:65:79:40
Unsupported card
```
```bash
# lsusb
Bus 003 Device 019: ID 1050:0407 Yubico.com Yubikey 4/5 OTP+U2F+CCID
```
```bash
ykman info
Device type: YubiKey 5 NFC
Serial number: 875640064
Firmware version: 5.6.0
Form factor: Keychain (USB-A)
Enabled USB interfaces: OTP, FIDO, CCID
NFC transport is disabled.

Applications
OTP         	Enabled
FIDO U2F    	Enabled
FIDO2       	Enabled
OATH        	Enabled
PIV         	Not available
OpenPGP     	Not available
YubiHSM Auth	Not available
```

Additionally, I'm trying to get this to work with Keepassxc, I'm not sure how to use `ykman` to put it in this mode:
[image]

polhenarejos, 2023-10-24:
I got the same in Windows. Strangely, Windows does not send GetFeature reports.

If you use `ykman -l TRAFFIC list` probably you will get a WinError 1: Incorrect function.

Analyzing with Wireshark, Windows does not even send GetFeature packet. Do not know why. Probably needs an additional parameter in the USB Descriptor.

polhenarejos, 2023-10-24:
If some of you has a Yubi4/5, `lusb -vvv` will be very helpful.

641i130, 2023-10-25:
The one I ordered just came today!
```
lsusb -v -d 1050:0407

Bus 003 Device 009: ID 1050:0407 Yubico.com Yubikey 4/5 OTP+U2F+CCID
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1050 Yubico.com
  idProduct          0x0407 Yubikey 4/5 OTP+U2F+CCID
  bcdDevice            5.43
  iManufacturer           1 Yubico
  iProduct                2 YubiKey OTP+FIDO+CCID
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0096
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower               30mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      71
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval              10
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
[... 78 lines cut ...]
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval              32
Device Status:     0x0000
  (Bus Powered)
```

polhenarejos, 2023-10-25:
Found the bug.

TinyUSB in Pico SDK does not set the FEATURE report for keyboard HID. Windows expects a value for FEATURE report to set the FeatureByteLength, which is 0 by default. When YKMAN asks for a 9-bytes of FEATURE report, Windows detects a mismatch between the request length (9) and the actual FeatureByteLength (0) and raises an exception.

I figured out how to inject the FEATURE report to HID descriptor and it works now on Windows. Hope it will not break anything in Linux/macOS.

641i130, 2023-10-25:
With the Yubikey or the pico-fido? Did you update the development branch or main branch?

polhenarejos, 2023-10-25:
With the Yubikey. I managed to make it work with Yubikey Manager and YKMAN but it does not with Yubikey Authenticator. As soon I get it fixed I'll push the changes to dev branch.

641i130, 2023-10-26:
```
Using reader with a card: Yubico YubiKey OTP+FIDO+CCID 00 00
3b:fd:13:00:00:81:31:fe:15:80:73:c0:21:c0:57:59:75:62:69:4b:65:79:40
Personal Identity Verification Card
```

polhenarejos, 2023-10-28:
Pushed latest changes. It should work in Windows too.

641i130, 2023-10-29:
```c
make
[  1%] Built target bs2_default
[  2%] Built target bs2_default_padded_checksummed_asm
[  3%] Performing build step for 'ELF2UF2Build'
[100%] Built target elf2uf2
[  3%] No install step for 'ELF2UF2Build'
[  3%] Completed 'ELF2UF2Build'
[  6%] Built target ELF2UF2Build
[  7%] Building C object CMakeFiles/pico_fido.dir/src/fido/fido.c.obj
/pico-fido/src/fido/fido.c: In function 'fido_ctor':
/pico-fido/src/fido/fido.c:70:18: error: passing argument 1 of 'register
_app' from incompatible pointer type [-Werror=incompatible-pointer-types]
   70 |     register_app(fido_select, fido_aid);
      |                  ^~~~~~~~~~~
      |                  |
      |                  int (*)(app_t *) {aka int (*)(struct app *)}
In file included from /pico-fido/src/fido/fido.c:20:
/pico-fido/pico-hsm-sdk/src/apdu.h:35:25: note: expected 'app_t * (*)(ap
p_t *, const uint8_t *, uint8_t)' {aka 'struct app * (*)(struct app *, const unsigned 
char *, unsigned char)'} but argument is of type 'int (*)(app_t *)' {aka 'int (*)(stru
ct app *)'}
   35 | extern int register_app(app_t *(*)(app_t *, const uint8_t *, uint8_t));
      |                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/pico-fido/src/fido/fido.c:70:5: error: too many arguments to function 
register_app'
   70 |     register_app(fido_select, fido_aid);
      |     ^~~~~~~~~~~~
/pico-fido/pico-hsm-sdk/src/apdu.h:35:12: note: declared here
   35 | extern int register_app(app_t *(*)(app_t *, const uint8_t *, uint8_t));
      |            ^~~~~~~~~~~~
cc1: all warnings being treated as errors
make[2]: *** [CMakeFiles/pico_fido.dir/build.make:76: CMakeFiles/pico_fido.dir/src/fid
o/fido.c.obj] Error 1
make[1]: *** [CMakeFiles/Makefile2:1509: CMakeFiles/pico_fido.dir/all] Error 2
make: *** [Makefile:91: all] Error 2
```
Most recent pull of `development`.

polhenarejos, 2023-10-30:
You have to update `pico-sdk-hsm` submodule too.

641i130, 2023-10-31:
[image]

I'm able to configure it correctly and detect it in Windows 11.

I believe whatever changes you made in the development branch fixed it!

c1fd5736f90dd9bbc999a192ce1d89bee8584ee5
c1fd5736f90dd9bbc999a192ce1d89bee8584ee5
c24be5a6319c98cc95ed410807d5c7679782cecf
da94a824877b46c2e4cc5c63a9375269a156075b
7bf26b28fcf14621f1613fb53734eadc10b4b89f
0fd36806cced45b11c1ed36c14fc505cd8d3d171

I think this issue can be closed then!

closed 2023-11-02 by polhenarejos