polhenarejos/pico-fido #13: issue with rp2040-zero not work after restart

alessiol78, 2023-07-20:
I try to flash firmare for rp2040-zero and work as "yubico neo" but when unplug from USB and reconnect not restart. As mentioned in [this forum](https://www.reddit.com/r/adafruit/comments/n47vac/feather_rp2040_not_working_after_restartafter/) problably can be solved setting a variable "This pico-sdk requires PICO_BOARD variable to be set to adafruit_feather_rp2040. By default, the sdk was using pico.h."
The yubico authenticator show "webauthn" and "authenticator" functions but in this case show a smartcard error connection:
[image]

polhenarejos, 2023-08-14:
You can also download a precompiled firmware for adafruit_feather_rp2040. Did you give it a try?

alessiol78, 2023-09-10:
> You can also download a precompiled firmware for adafruit_feather_rp2040. Did you give it a try?

I try adafruit_feather_rp2040 and solve the restart issue. Only a note: the LED not work at all.

closed 2023-11-02 by polhenarejos