polhenarejos/pico-fido #15: Ability to change pins for LED and Button

cortices, 2023-08-15:
I'm trying to work out where the Hardware Abstraction Layer for choosing the button and LED pins lives, as I wish to break out both to external IO pins and use those elsewhere, and the Pi Pico board doesn't break out BOOTSEL or LED (GP25).

Which file or dependency are these defined in? I couldn't find it in this repo or the hsm_sdk one either.

polhenarejos, 2023-08-16:
LED is in https://github.com/polhenarejos/pico-hsm-sdk/blob/main/src/main.c#L197
Button task is handled by TinyUSB, already embedded to Pico SDK.

Usually it takes the macros defined by `PICO_BOARD` when compiling, that defines `PICO_DEFAULT_LED_PIN` and others. If you want to use another board, you have to specify by `PICO_BOARD`.
On the contrary, if you are using your own board, you should add the corresponding header file at Pico SDK (not the Pico HSM SDK) defining LED pins, capacity of flash memory and other parameters.

cortices, 2023-09-16:
I added some `#undefine`s to replace `PICO_DEFAULT_LED_PIN` and `PICO_DEFAULT_LED_PIN_INVERTED`, and replaced calls to `board_button_read()` with my own function. This was sufficient.

closed 2023-09-16 by cortices