polhenarejos/pico-fido #16: Cannot generate ssh key

aa889788, 2023-09-17:
I'm trying to auth the ssh using pico-fido, but the key cannot be generated
```bash
$ ssh-keygen -vvvv -t ecdsa-sk
Generating public/private ecdsa-sk key pair.
You may need to touch your authenticator to authorize key generation.
debug3: start_helper: started pid=14873
debug3: ssh_msg_send: type 5
debug3: ssh_msg_recv entering
debug1: start_helper: starting /usr/lib/ssh/ssh-sk-helper 
debug1: sshsk_enroll: provider "internal", device "(null)", application "ssh:", userid "(null)", flags 0x01, challenge len 0
debug1: sshsk_enroll: using random challenge
debug1: sk_probe: 1 device(s) detected
debug1: sk_probe: selecting sk by touch
debug1: ssh_sk_enroll: using device /dev/hidraw2
debug1: ssh_sk_enroll: fido_dev_make_cred: FIDO_ERR_RX
debug1: sshsk_enroll: provider "internal" failure -1
debug1: ssh-sk-helper: Enrollment failed: invalid format
debug1: main: reply len 8
debug3: ssh_msg_send: type 5
debug1: client_converse: helper returned error -4
debug3: reap_helper: pid=14873
Key enrollment failed: invalid format
```
```bash
$ ssh-keygen -vvvv -t ecdsa-sk -O resident
Generating public/private ecdsa-sk key pair.
You may need to touch your authenticator to authorize key generation.
debug3: start_helper: started pid=14953
debug3: ssh_msg_send: type 5
debug3: ssh_msg_recv entering
debug1: start_helper: starting /usr/lib/ssh/ssh-sk-helper 
debug1: sshsk_enroll: provider "internal", device "(null)", application "ssh:", userid "(null)", flags 0x21, challenge len 0
debug1: sshsk_enroll: using random challenge
debug1: sk_probe: 1 device(s) detected
debug1: sk_probe: selecting sk by touch
debug1: ssh_sk_enroll: using device /dev/hidraw2
debug1: check_sk_options: option uv is unknown
debug1: key_lookup: fido_dev_get_assert: FIDO_ERR_RX
debug1: ssh_sk_enroll: key_lookup failed
debug1: sshsk_enroll: provider "internal" failure -1
debug1: ssh-sk-helper: Enrollment failed: invalid format
debug1: main: reply len 8
debug3: ssh_msg_send: type 5
debug1: client_converse: helper returned error -4
debug3: reap_helper: pid=14953
```

aa889788, 2023-09-17:
I don't know if it's because the pico-fido does not support ssh, or my environment has some issues?

polhenarejos, 2023-09-17:
Which OS?
Is you pico-fido properly recognized?

polhenarejos, 2023-09-17:
For first error (no resident), it is caused due to PIN protection. For an unknown reason, ssh-keygen does not prompt for PIN.
If you remove PIN protection, it should work.

Second error is related to resident list. I'll take a look.

polhenarejos, 2023-09-17:
Digging into ssh-keygen code, it seems that all problems are related with PIN, which is not asked. Need to fix it first.

commit 9c90095e (2023-09-17): CBOR errors are not sent through CTAPHID_ERROR command, but in CBOR response instead. Fixes #16
closed 2023-09-18 by polhenarejos