polhenarejos/pico-fido #17: No LED Activity.

AndrewAubury, 2023-09-17:
I am using a Raspberry Pi Pico W, flashed with pico_fido_pico_w-5.4.yubikey5.uf2 (So Version 5.4 as a Yubikey5).

Upon plugging in and registering it, at no point does the built in light flash on.

Is there something I am missing?

polhenarejos, 2023-09-18:
Need more info. 
- Does it even blink?
- Does reset fix it? 
- What are you doing to stop blinking? 
- Is there any particular pattern to reproduce it?

polhenarejos, 2023-09-18:
Also, can you try with latest version 5.6 released today?

AndrewAubury, 2023-09-18:
> Does it even blink?
No there is no LED activity at all, it does not turn on.

> Does reset fix it?
Unfortunately not, I have tested 2 Pico W's brand new from the store I used Pico Nuke before install.

> What are you doing to stop blinking?
N/A as there is no activity at all.

> Is there any particular pattern to reproduce it?
N/A as there is no activity at all.

>  Version 5.6
I had issues getting 5.6 to work, and dropped down to 5.4. The issue I was having with 5.6 was after Windows Security asked and set a pin it would then not accept the pin and made the key useless however 5.4 worked without issue.

It is worth noting that the issue with the LED not lighting is present on my Pico Ws on BOTH versions

polhenarejos, 2023-09-18:
When did you try with 5.6? I detected a bug in that release which made me make a new release, uploaded few minutes ago. Is the same you tried?

AndrewAubury, 2023-09-18:
I had tried it last night, I have now reflashed it with 5.6 the LED issue is still present however I can use the key as intended.

polhenarejos, 2023-09-18:
I'have uploaded a new firmware only for pico_w. It should work.

closed 2023-09-18 by polhenarejos

AndrewAubury, 2023-09-18:
Uploaded file is too large - Unable to use Pico Patcher

polhenarejos, 2023-09-18:
Fixed. Increased max size for Pico Patcher.

DevSnyco, 2024-08-19:
Does it even blink?
- No, never blinked after flash.
Does reset fix it?
- No, I've tried resetting it, this is a fresh batch and it doesn't work on both of my pico pi zeros
What are you doing to stop blinking?
- Nothing, it never starts blinking after flash.
Is there any particular pattern to reproduce it?
- Just moved the u2f file into the Pi.

Note: The Pico works perfectly fine, the LED is the only thing that isn't working.

polhenarejos, 2024-08-19:
Which board?

DevSnyco, 2024-08-19:
> Which board?

RP2040 Pi Zero (Same board used in the screenshots)

polhenarejos, 2024-08-19:
Put here the website with the specs of the board (or where to buy it) and I'll take a look.

The board used in the screenshots is the Tiny2040 from Pimoroni, probably not the one you mention.

DevSnyco, 2024-08-19:
https://my.shp.ee/3oMHRpc

polhenarejos, 2024-08-19:
I must register in that web but I need a Taiwan phone number to do so. Is there any other way to see the product without being registered?

DevSnyco, 2024-08-19:
i can send screenshots instead, if you'd like
or maybe a html file of what i see?

polhenarejos, 2024-08-19:
Yes, a screenshot is fine.

DevSnyco, 2024-08-19:
[image]
[image]

DevSnyco, 2024-08-19:
no, im using the `pico_fido_pico-5.10.uf2` firmware, it works perfectly fine, LED is the only thing not working

polhenarejos, 2024-08-19:
You must use the one for your board. Not all boards are the same.

Try with this: https://github.com/polhenarejos/pico-fido/releases/download/v5.10/pico_fido_waveshare_rp2040_zero-5.10.uf2

polhenarejos, 2024-08-19:
It should work, as the key space is different from the firmware space.

DevSnyco, 2024-08-19:
i have just installed it, it works perfectly well now! thank you very much!

commit 6c85d574 (2023-09-18): Added support for LED in Pico W.

Fixed #17.
commit 041bb788 (2023-11-21): Added support for LED in Pico W.

Fixed #17.