polhenarejos/pico-fido #18: Unable to register FIDO2 credentials.

ArykDev, 2023-09-20:
I have tested my Raspberry Pi Pico with `https://webauthn.io`, but it just isn't working.

[image]

I have already patched the .uf2 file with `234b:000`, opensc-tool reports "No cards found" as well.

dmesg log:
```
[11890.475945] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[11890.475952] usb 1-1: Product: Pico Key
[11890.475957] usb 1-1: Manufacturer: Pol Henarejos
[11890.475961] usb 1-1: SerialNumber: E6614C30930D4C2C
[11890.483091] hid-generic 0003:234B:0000.0009: hiddev96,hidraw0: USB HID v1.11 Device [Pol Henarejos Pico Key] on usb-0000:00:14.0-1/input0
[11890.486215] input: Pol Henarejos Pico Key as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.1/0003:234B:0000.000A/input/input20
[11890.540834] hid-generic 0003:234B:0000.000A: input,hidraw1: USB HID v1.11 Keyboard [Pol Henarejos Pico Key] on usb-0000:00:14.0-1/input1
```

lsusb output (for this device):
`Bus 001 Device 021: ID 234b:0000 Free Software Initiative of Japan Gnuk Token`

It is definitely seeing the device, however it's just not able to register credentials for some reason.

ArykDev, 2023-09-20:
> Firefox has a flaky support to FIDO2. Can you try it with other browser like Chrome?
> 
> Also, paste here what `opensc-tool -an` reports.

```
No smart card readers found.
Failed to connect to reader: No readers found
```

Will try using this in Chrome, thanks.

ArykDev, 2023-09-20:
Works perfectly fine in Chrome, strange.
Any ideas on how I can get this working in Firefox?

ArykDev, 2023-09-20:
So I've narrowed this down to being able to authenticate with this, however oddly, I am unable to register, I just get a random "unexpected error".
Will try to find something about this.

polhenarejos, 2023-11-21:
Version 5.8 is released. Can you try if it fixes this error?

ArykDev, 2023-12-11:
> Version 5.8 is released. Can you try if it fixes this error?

Hey, just saw this, absolutely, I'll give it a go now.

closed 2023-12-11 by jackeilles

ArykDev, 2023-12-11:
@polhenarejos Working perfectly now under Windows 11 running Firefox as well as with KDE on Xenia Linux running Firefox.
Closing issue.

cortices, 2024-03-31:
I still had the issue persisting, but then I tried the fix suggested in the debugging of this Firefox bug https://bugzilla.mozilla.org/show_bug.cgi?id=1859425:
Setting `security.webauthn.ctap2 = false` in Firefox's about:config page.

This instantly fixed the issue on webauthn.io for me.

jackeilles, 2024-05-02:
Oh, interesting. Was it still happening after the supposed fix?