polhenarejos/pico-fido #19: Challenge Response Failed to connect

641i130, 2023-10-21:
Using `ykman` in this reddit post:
https://www.reddit.com/r/KeePass/comments/opx34q/keepassxc_and_yubikeys_setting_up_the/
I'm getting the follow response from the device:
```bash
# ykman otp chalresp -t -g 2
Using a randomly generated key (hex): 4e613cf377db2a67bfeebe9ba8b70d99f90daa99
Program a challenge-response credential in slot 2? [y/N]: y
ERROR: Failed to write to the YubiKey. Make sure the device does not have restricted access (see "ykman otp --help" for more info).
```
I'm using a Xiao Seeed if it matters.

polhenarejos, 2023-10-24:
Actually, the error comes from the same as in #12: Windows is not able to HID GET FEATURE. But whilst in #12 ykman is able to fallback to another backend, this is not the case for OTP, which only uses HID FEATURE report backend. I cannot figure out how to make it work in windows.

641i130, 2023-10-31:
Testing it with the newest pull, I get past the Failed to write error this time, but it seems Keepassxc doesn't detect it.
[image]

[image]

Progress of course, love the project man, keep it up! This could become big when it does work.

641i130, 2023-10-31:
I'm able to get it to generate, but keepassxc doesn't seem to see it yet oddly.
[image]
[image]

Heres what it should look like:
[image]

I'd like to help debug this, but I'm not sure what information is needed. Maybe USB pcap?

polhenarejos, 2023-10-31:
Which OS?

641i130, 2023-10-31:
Arch Linux

polhenarejos, 2023-10-31:
Fixed in e5d1ef2.

Do not forget to pull sdk too. It works with KeepassXC when a slot is properly configured with Challenge-response credential type (via Yubikey Manager for instance.

641i130, 2023-11-01:
It works! Thank you so much for making this project. It's very interesting to me, someone getting into embedded systems design! I also like that we can make our own keys instead of buying them all from Yubico!!!

commit e5d1ef29 (2023-10-31): Fixed OTP read packet through HID interfaces.

Fixes #19.
commit ba57cc45 (2023-11-21): Fixed OTP read packet through HID interfaces.

Fixes #19.
closed 2023-11-01 by 641i130