polhenarejos/pico-fido #20: Can't use it to passwordless login to Google

Le55mmore, 2023-10-25:
I flashed last firmware to generic rp2040 and it works for 2fa when i need to enter my password and press a button, but every time i try to setup passwordless login, it asks and sets pin, and send credentials, everything seems fine, but when i try to login using passwordless method it accepts pin as correct but seems not find credentials, don't ask to press button and asks for different key. When i try to do same in yubikey playground it works perfectly.
[image]
[image]
[image]
I just don't get it. Should it work?

polhenarejos, 2023-10-25:
Which browser are you using?
Can you try with www.webauthn.io?

Le55mmore, 2023-10-25:
I use brave(chromium)
https://webauthn.io/ works fine, i also checked github it works too.
When i removed token for password+button from google and now it does not request pin and just waiting for something
[image]
[image]
same in chrome, edge. Firefox asks for button press first, then asks for pin and tells to use different key device.

aleytus, 2023-12-02:
@Le55mmore В настройках безопасности гугла выбери пункт "Ключи доступа" - "Создать ключ доступа" - "Использовать другое устройство"

PS. Использую последний фаерфокс и ltsb сборку 10 винды
Свой собственный сертификат и VID/PID (FEFF:FCFD)
[image]
[image]

Le55mmore, 2023-12-05:
I nuked flash and updated to 5.8, passwordless works now everywhere. Cool

closed 2023-12-05 by Le55mmore