polhenarejos/pico-fido #27: [Feature Request] - NFC Support

641i130, 2023-11-10:
Not too sure I fully understand how fido keys work over NFC (13.56 MHz) but I did find this NFC rp2040 dev board that might be a good starting point for getting this functional:
https://thepihut.com/products/challenger-rp2040-nfc-with-antenna
I guess you would also need an NFC reader connecting to the computer to test it as well. Here are the ones that people with yubikeys say work:
- https://www.amazon.com/Omnikey-HID-5022-Contactless-Reader/dp/B079T2FKN1
- https://www.amazon.com/dp/B01KIKBYAG

I think its anything with support for ISO 14443-4 (NFC-A) support, from my understanding.

wzqvip, 2023-12-21:
nrf52832/840 should be better for this purpose. Like the Charmeleon.

DevSnyco, 2024-08-11:
yes, but can u make it so nfc works when its plugged into power? because if it support nfc, then it supports ios and android too

DevSnyco, 2024-08-19:
also, how would the nfc work on yubikeys then?

wzqvip, 2024-08-26:
> also, how would the nfc work on yubikeys then?

That's a different chip, and possibly could be powered from nfc.

wzqvip, 2024-08-26:
> The nrf52840 can't be supplied from NFC, it needs additional power, which makes it pretty useless as a wireless FIDO2 key.

Maybe it's fine if you connect it to pc and then tab with phone to auth. )

DevSnyco, 2024-08-31:
> > The nrf52840 can't be supplied from NFC, it needs additional power, which makes it pretty useless as a wireless FIDO2 key.
> 
> Maybe it's fine if you connect it to pc and then tab with phone to auth. )

ok, then can we get that feature then?

closed 2025-01-15 by polhenarejos