polhenarejos/pico-fido #29: Static password does not take into account the length

sylvainpelissier, 2023-11-16:
It seems for static password the length is limited to 8 characters:
```bash
$ ykman otp static  1
Enter a static password: bbbbbbbbbbbbbbbbbbbbbbbbbbbbbb
Slot 1 is already configured. Overwrite configuration? [y/N]: y
```
Then pressing the button gives only `bbbbbbbb`

The same happens with password generation:
```bash
$ ykman otp static --length 24 --generate 1
```

polhenarejos, 2023-11-16:
It is fixed but `ykman` sets the `SHORT_TICKET` flag, which implies only the half of the password is sent. Is this happening with a Yubikey too? To me is quite strange.

sylvainpelissier, 2023-11-18:
For the static password mode, the full programmed password is returned. I think this option is used only for Yubico OTP: https://developers.yubico.com/yubikey-personalization/Manuals/ykpersonalize.1.html

polhenarejos, 2023-11-20:
From the ykpersonalize webpage:
```
short-ticket | Limit the length of the static string to max 16 digits. This flag only makes sense with the -ostatic-ticket option. When -oshort-ticket is used without -ostatic-ticket it will program the YubiKey in "scan-code mode", in this mode the key sends the contents of fixed, uid and key as raw keyboard scancodes. For example, by using the fixed string h:8b080f0f122c9a12150f079e in this mode it will send Hello World! on a qwerty keyboard. This mode sends raw scan codes, so output will differ between keyboard layouts.
```
and YKMAN sets by default SHORT_TICKET flag:
https://github.com/Yubico/yubikey-manager/blob/51a7ae438c923189788a1e31d3de18d452131942/yubikit/yubiotp.py#L474

```
        self._update_flags(CFGFLAG.SHORT_TICKET, True)
```
It does not set STATIC_TICKET, so I guess it is used as raw scancode mode.

When you do `$ ykman otp static --length 24 --generate 1`, does it return 24 chars or 12 in a Yubikey?

sylvainpelissier, 2023-11-21:
It returns a 24 characters password when pressing the Yubikey.

closed 2023-11-21 by polhenarejos

polhenarejos, 2023-11-21:
Fixed in Release 5.8

commit 96de6efe (2023-11-16): OTP static passwords are 38 bytes length.

A static password uses fixed, uid and key fields (sum 38). However, Yubikey sets short_ticket flag which implies the half of the password is sent.

Fixes #29.
commit e757ad29 (2023-11-21): Removing SHORT_TICKET limitation.

It is not used to return the half of ticket, but to combine with static to produce hex scancodes.

Fixes #29.