polhenarejos/pico-fido #30: Yubikey 5 doesn't work for yubikey manager on windows 10

JakeFrosty, 2023-11-20:
All seems to be working except any attempts to modify the device in windows 10, It always shows up as "Yubikey" and Firmware 3.0.0, despite the fact that I flashed pico_fido_pico-5.6.yubikey5.uf2 (using the web patcher).

[image]

My pico model, just a cheap copy but with USB-C and an RGB LED light in the back
[image]

Hoping to get some help getting this to work, I wanted to use this as a primary/backup alongside my Yubico Security Key

polhenarejos, 2023-11-20:
I found a bug in 5.6 which caused you mention not working in Windows. It is fixed in the `development` branch but I have not made a public release yet. I am waiting to fix a bug with OTP and static passwords. If you have installed Pico toolchain and SDK, you can build the development branch; otherwise, as soon I fix it I'll release the 5.8 which fixes your bug. 

If you have a yubikey, in the meanwhile, can you take a look on #29 and see what returns when `SHORT_TICKET` is enabled? Thanks.

JakeFrosty, 2023-11-20:
I'm interested in helping, however I'm a complete newbie. May I know how I can go about this SHORT_TICKET problem? I have a Yubikey Security Key (Blue) that can do the FIDO2 only and not OTP or PIV
[image]

polhenarejos, 2023-11-20:
Mainly you have to take your Yubikey and create a short ticket via `ykman otp static --length 24 --generate 1`. Then, see which is the length returned when you push the button, either `12` or `24` characters.

JakeFrosty, 2023-11-20:
Unfortunately my Yubikey does not support OTP, only FIDO2

closed 2023-11-21 by polhenarejos

polhenarejos, 2023-11-21:
Fixed in Release 5.8

commit 1ce0d98c (2023-11-21): OTP callbacks must be initialized on ctor.

Fixes #30.