polhenarejos/pico-fido #33: Nitrokey 3 secrets list name display error

zvrh, 2023-12-12:
## secrets list name display error：

[image]

## after the following modification, it will work normally：

https://github.com/polhenarejos/pico-fido/blob/7e2ecdbc5659fb4de39b848503d759e39d85d573/src/fido/oath.c#L290-L293

[image]

polhenarejos, 2024-01-08:
Hi. Do you know why is that? If we apply your modification, probably it won't work with Yubikey. Perhaps is a bug with `nitropy` app?

zvrh, 2024-01-08:
I don't quite understand this issue either. As you said, maybe this is a problem with Nitrokey APP.

closed 2024-01-08 by zvrh