polhenarejos/pico-fido #34: Which part of the dongle i made defined the "key"?

ryleenhughes, 2023-12-25:
Hello, thank you for making this wonderful project! It works really nice, and I love it so much!  
I apologize for my stupid question, however I don't have enough capabilities to read complicated code, so I think it would be nice if you got spare time to simply answer this:  

1. Considering the hardware quality of dev-boards isn't always as good as real FIDO devices, I wonder which part that really defined the "key" (aka the different part from other pico-fido dongle I make)? Is it the rp2040 board itself defined the key (like unchangeable serial number or something that included the rp2040 chip), or the key is included in the UF2 file?  

2. In another word, if I meant to make two same dongles that can replace each other (just for example, I won't use it maliciously), would it make two dongles totally same if I put same uf2 file to both of the dongle?  

3. Are there way to backup the "key" in clear text so I can backup it on a paper instead of saving some files? 

(I'm not worrying that I leaked the "key" to hackers or something, since my account is not important at all. All I'm worried is that someday the hardware broken, and I lost access to my account, that's why I need to making sure I know a correct way to backup my "key")

Thank you in advance for answering my stupid question, and have a nice day!

polhenarejos, 2024-01-08:
Hi.
1. I am not sure understanding your question. There are no keys within UF2 file. All keys are generated and stored in the flash memory of your pico board. Keys are fed with the HRNG of rp2040 ensuring all are completely random. There is no mechanism like master seed and derivation.
2. No. Two dongles generate two different keys. 
3. Following Fido Alliance guide, there should not be any mechanism to backup or export the key outside the dongle. 

The correct way to backup your credentials is to register a secondary dongle to the same account. In case that you cannot login with your primary device (due to either lost or broken device), you can still login with your secondary device to be able to register a replacement dongle. Therefore, there should be always 2 dongles registered at your account.

ryleenhughes, 2024-01-08:
Thank you for your response!

> Following Fido Alliance guide, there should not be any mechanism to backup or export the key outside the dongle.   

The concern I have is that, regardless of how many dongles I register, the hardware is simply not reliable. Therefore, I am wondering if I can back up the keys and create a new dongle at any time if all the hardware dongles were broken or lost.

I noticed in [another issue](https://github.com/polhenarejos/pico-fido/issues/35) that mentions the possibility of dumping the entire SPI flash to back up the keys. Considering your statement that it's not possible, I assume what they mentioned is not recommended or reliable?

I apologize for another dumb question, but can I reset a dongle? For example, if I no longer need the old key and want a brand-new fido dongle?

Thank you so much!

polhenarejos, 2024-01-08:
This is a particularity of rp2040 boards. Since rp2040 does not have an integrated flash memory, it always requires an external one and SPI bus is not encrypted, exposing the contents of flash memory. 
With `pico-tool` from RPi repo, the entire flash memory can be dumped and restored in another board. The dumped file is always encrypted. I tried to follow Fido guides as much as possible given the rp2040 hw, but the possibility of dumping the flash memory is always there and I cannot disable it.

If you want to reset a dongle, you can use `pico-nuke`. Just drap and drop the nuke UF2 and you'll have a blank factory pico board. See https://github.com/polhenarejos/pico-nuke for binaries.

polhenarejos, 2024-01-08:
Actually, since Fido Alliance do not recommend to implement any backup system, I won't do that. In case you need a backup/restore keychain system, you can take a look to [Pico HSM](https://github.com/polhenarejos/pico-hsm).

ryleenhughes, 2024-01-08:
got it, Thank you so much for all the thing!

closed 2024-01-08 by ryleenhughes