polhenarejos/pico-fido #35: Questions about upgrade firmware and backup

lifetyper, 2023-12-29:
Hi,
Firstly, let me express my gratitude for this project.I've got some questions about firmware upgrade and backup.

1.If I upgrade the firmware by BOOTSEL(the same way as initial firmware download),what happens to the private keys/master key stored in the spi flash?Will them be keeped or everything  just wiped out?

2.I'm using ver 5.8 firmware here but I see there is an ver 5.8 Eddsa  firmware,can I upgrade my key with this Eddsa version directly?

3.I understand that the proper way to backup a key is register a second key as backup.

But what if I save the content of spi flash by picotool or even just dump the whole flash content by hardware flash programmer and re-program it into another hardware.Can I get a duplicated key by doing this?

Thanks!

lifetyper, 2023-12-29:
OK,let me answer my first two quesions.
1.upgrade firmware won't delete any keys stored in the hardware.
2.upgrade from ver 5.8 to ver 5.8 Eddsa is also safe.

lifetyper, 2024-01-03:
just confirmed that duplicate the content of spi flash == duplicate key

closed 2024-01-03 by lifetyper

ryleenhughes, 2024-01-03:
Hello, thank you so much for this research! Would you mind share that how do you duplicate the content of spi flash? I just wanted to make sure i did everything right.

lifetyper, 2024-01-03:
> Hello, thank you so much for this research! Would you mind share that how do you duplicate the content of spi flash? I just wanted to make sure i did everything right.

Hi,in fact it's quite simple. Using [picotool](https://github.com/raspberrypi/picotool) to dump the spi flash and flash it into a new pico key will done.
On old key:
`picotool save -a dump.bin`
On new key:
`picotool load dump.bin`

There're also some other stuffs about bootsel,replace usb driver or even compile picotool.I wrote a simple guide here FYR.

You are here: [Home](https://lifetyper.com/) [misc](https://lifetyper.com/category/misc.html) How to clone a Pico Key（Pico Fido Security Key） 

[How to clone a Pico Key](https://lifetyper.com/posts/how_to_clone_a_pico_key_en.html)

ryleenhughes, 2024-01-03:
@lifetyper Thank you so much for the info! Have a nice day! :)