polhenarejos/pico-fido #37: Issue in Pico-fido-tool.py

Aadinthosh, 2024-02-21:
I am trying to use pico-fido-tool.py file for accessing the pico-fido board with the pin and while doing so I am getting this error message
**AttributeError: 'NoneType' object has no attribute 'capabilities'**

polhenarejos, 2024-02-21:
I'd need more debug, like the entire traceback, which firmware version, etc.

Aadinthosh, 2024-02-21:
@polhenarejos 
I am using the firmware version  pico_fido_pico-5.8.uf2  and I am using the Pico H board for experimentation. The exception traceback is mentioned below

 D:\IoT-Projects\pico-fido\tools> python3 .\pico-fido-tool.py -p 12345678
Pico Fido Tool v1.6
Author: Pol Henarejos
Report bugs to https://github.com/polhenarejos/pico-fido/issues

Traceback (most recent call last):
  File "D:\IoT-Projects\pico-fido\tools\pico-fido-tool.py", line 469, in <module>
    run()
  File "D:\IoT-Projects\pico-fido\tools\pico-fido-tool.py", line 466, in run
    main(args)
  File "D:\IoT-Projects\pico-fido\tools\pico-fido-tool.py", line 452, in main
    ctap = Ctap2Vendor(dev)
  File "D:\IoT-Projects\pico-fido\tools\pico-fido-tool.py", line 111, in __init__
    super().__init__(device=device, strict_cbor=strict_cbor)
  File "C:\Users\santh\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\fido2\ctap2\base.py", line 204, in __init__
    if not device.capabilities & CAPABILITY.CBOR:
AttributeError: 'NoneType' object has no attribute 'capabilities'

polhenarejos, 2024-02-21:
I can reproduce it in Windows but not in macOS nor in Linux. Seems it is a problem with `fido2` library and Windows.

Need deeper debug.

polhenarejos, 2024-02-21:
False alarm. FIDO2 devices in Windows need Admin privileges. Try to open a new Console with Admin privileges and call it again. It should work.

closed 2024-02-21 by polhenarejos