polhenarejos/pico-fido #38: Fail to open otp on windows client

agczsz, 2024-03-03:
Fail to open otp on windows client，it can only use on my phone

polhenarejos, 2024-03-07:
More details needed:
- OS
- Client software
- FIDO version
- Error message

agczsz, 2024-03-09:
waveshare-rp2040-one,windows,totp
[image]
other function is okey
[image]
and totp works well on android with otg

fucksophie, 2024-05-09:
TOTP works fine on Linux. Latest (as of posting) yubico authenicator, on
```
╭──────────────────┬────────────────────────────────────────────────────────────╮
│ kernel-name      │ Linux                                                      │
│ nodename         │ constellation                                              │
│ kernel-release   │ 6.8.8-zen1-1-zen                                           │
│ kernel-version   │ #1 ZEN SMP PREEMPT_DYNAMIC Sun, 28 Apr 2024 15:59:30 +0000 │
│ machine          │ x86_64                                                     │
│ operating-system │ GNU/Linux                                                  │
╰──────────────────┴────────────────────────────────────────────────────────────╯
```

polhenarejos, 2024-07-20:
v5.10 introduced some bugfixes for Windows. Reopen if it still persists.

closed 2024-07-20 by polhenarejos