polhenarejos/pico-fido #41: Pico-Fido on Android Smartphone

seeers, 2024-04-04:
Do I have to consider anything if I want to use the pico fido on an Android smartphone? 
Unfortunately, it is not recognised in my case. Neither with the original VID/PID nor with Yubikey 4/5 OTP+FIDO+CCID patched. 
Do I have to pay attention to anything or is it generally not possible?

polhenarejos, 2024-04-05:
I did not test it with any Android phone. Did you try it with a genuine Yubikey?

seeers, 2024-04-10:
Unfortunately I don't have a Yubikey, I was hoping that someone here had already tried it

whitelynx, 2024-05-07:
It didn't work for me either on my phone initially - the board didn't even light up. I did get it to work by going into the phone settings, and turning on `OTG storage` in the System section. It seems like that setting turns off after 10 minutes, though, and it's a pain to have to go to settings every time you need to use your key. I don't have a Yubikey to compare to, sadly.

Phone: OnePlus 3T
Android version: 9
Pico Fido USB info: `idVendor`=1050, `idProduct`=0407

fucksophie, 2024-05-09:
Works perfectly on my phone. Absolutely no issues I ran into, worked out of the box.
 
 Phone: Moto G53
 Android version: 16
 USB info: `Bus 003 Device 003: ID 1050:0407 Yubico.com Yubikey 4/5 OTP+U2F+CCID`

fucksophie, 2024-12-17:
> it would be nice to have a android OTP software for pico-fido like https://github.com/beemdevelopment/Aegis/issues/336

Just use Yubikey Authenicator and go to the TOTP section?

fucksophie, 2024-12-17:
> but that is not opensourced.

https://github.com/Yubico/yubioath-flutter ?

polhenarejos, 2024-12-17:
The application passes a timestamp reference as a challenge.

omerakgoz34, 2024-12-21:
Passkey and security key functions work on my win11 but it doesn't work on my phone for some reason. My phone detects it as a keyboard but doesn't detect as a passkey or anything. But somehow yubikey auth android app detects it and storing 2FA code function works just fine. I even deleted the app and reinstalled to see if the codes actually stored on the picokey and yes it still works. But as the passkey function, yubikey app doesn't detect it as a passkey device.

Phone: Xiaomi 12T Android 14 with pico-fido v6.0 patched for yubikey5

Yubikey manager results:
[image]

polhenarejos, 2024-12-21:
Yubikey app also checks `usbProductName` (not sure if this is your case as I don’t have an Android phone). In v6.2 you will be able to commission a product name through the commissioner. 
If you don’t want to wait but test it you can try the nightly development build. Once you upload the latest development build go to Pico Commissioner and introduce “Yubico Yubikey” in the product name and use the VIDPID of Yubikey 4/5. After rebooting the key it should be detected. At least it’s how it works with the PC version.

closed 2025-01-15 by polhenarejos