polhenarejos/pico-fido #43: Version 5.6+ not working with lbuchs/WebAuthn library

lphooge, 2024-05-10:
For testing I ordered both a pimoroni tiny2040 and a waveshare Rp2040 zero.

After accidentially flashing version 2.2 to the tiny and the latest 5.8 to the waveshare, the tiny worked on my app, the waveshare did not (LED flashing but no function).

After further inspection I found its not tied to the hardware, both behave the same when flashed to the same version: 
- Its not just my app, it also does not work with the example for the library im using:
https://webauthn.lubu.ch/_test/client.html
- It DOES work with Version 5.4 and before (therefore I suspect the problem beeing with pico-fido and not the library)
- It DOES work with https://webauthn.io/ with all versions I tested
- Strangly, after using the key once with webauth.io, it kind of works with the other library but behaves eratically, sometimes not authenticating, sometimes asking for a pin despite userVerification=discouraged and sometimes crashing outright.

I have not yet found out what exactly triggers the different behavior, but suspect it to be in some kind tied to the pin function.

I did try nuking the flash, this did reset the "priming" via webauthn.io but did not fix the overall issue.

polhenarejos, 2024-05-13:
To summarize, it seems version 5.8 does not work with https://webauthn.lubu.ch/_test/client.html but it works with https://webauthn.io/  ?

PIN is management is performed by the current client, not by the Pico Fido. So, if it is erratic, it could be a client problem, specially if it works in one web but not in another. I'll debug with lubu's one.

In the meantime, can you try with https://demo.yubico.com/webauthn-technical/registration ?

polhenarejos, 2024-05-14:
I tested https://webauthn.lubu.ch/_test/client.html and it works smoothly. The only thing I saw is the Pico device has to be already set PIN. If it is a fresh new installation, you need first to set a PIN through webauthn.io and then it will work with lubu.ch, even if PIN is discouraged or required.

polhenarejos, 2024-07-20:
Probably fixed in v5.10

closed 2024-07-20 by polhenarejos