polhenarejos/pico-fido #44: Display Module for Raspberry Pi Pico + Implementation security

gpsa, 2024-05-18:
Do you think that adding a "Display Module for Raspberry Pi Pico" would solve the problem of having to keep the encryption key plain text?

If so, would you be able to point out which part of the code would require a change to wait on user's input?

gpsa, 2024-05-30:
Gotcha! Would you mind pointing where in the code we might need changes to do this?

polhenarejos, 2024-05-30:
Function `board_button_read()` at https://github.com/polhenarejos/pico-keys-sdk/blob/e4e43dae04d093286bd532d7486b24e8237b4960/src/main.c#L206 is the one that returns `true` or `false` when BOOTSEL button is pressed or release. You could override it by taking the prompt from display.

closed 2025-01-15 by polhenarejos