polhenarejos/pico-fido #46: Touch security key to continue...

fastchain, 2024-06-05:
Hello, 
I've just flashed RPI PICO 2020 (c) 
With this firmware 
https://github.com/polhenarejos/pico-fido/releases/download/v5.8/pico_fido_pico-5.8.uf2

after pico reboot  I see

```
[1466535.469647] usb 1-8: new full-speed USB device number 29 using xhci_hcd
[1466535.611275] usb 1-8: config 1 interface 2 altsetting 0 endpoint 0x82 has an invalid bInterval 0, changing to 10
[1466535.612120] usb 1-8: New USB device found, idVendor=feff, idProduct=fcfd, bcdDevice= 5.00
[1466535.612130] usb 1-8: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[1466535.612135] usb 1-8: Product: Pico Key
[1466535.612139] usb 1-8: Manufacturer: Pol Henarejos
[1466535.612142] usb 1-8: SerialNumber: E6612483CB1F932D
[1466535.618983] hid-generic 0003:FEFF:FCFD.0011: hiddev100,hidraw10: USB HID v1.11 Device [Pol Henarejos Pico Key] on usb-0000:00:14.0-8/input0
[1466535.620845] input: Pol Henarejos Pico Key as /devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.1/0003:FEFF:FCFD.0012/input/input36
[1466535.677164] hid-generic 0003:FEFF:FCFD.0012: input,hidraw11: USB HID v1.11 Keyboard [Pol Henarejos Pico Key] on usb-0000:00:14.0-8/input1

```

in dmesg  (I use Linux box 6.8.7-arch1-1) and according to green led it's in "Active mode".

And when I go to https://webauthn.io/ to run registration test, I get "Touch security key to continue".  I push BOOTSEL button, but nothing happens. What am I doing wrong?

Thank you!

polhenarejos, 2024-06-05:
Which browser? With Chrome should work out the box, with Firefox works partially and some parameter needs to be tuned in config params. Other browsers have not been tested yet.

fastchain, 2024-06-05:
hi!

> Which browser?
FF 126 as snap package 

And I moved a bit forward:

I've patched the firmware to "Yubikey 4/5 OTP+FIDO+CCID" with https://www.picokeys.com/pico-patcher/ . This made FF "to see" the token, and now FF asks for a pin on "registration".   Is it some default PIN?

Since I don't know the pin, I reset the pin with ykman tool

```
ykman fido access change-pin  --new-pin 000000
```

And when I provide a new pin to FF, it accepts it  asks to push the button.  After the push I see this 

[image]

>Firefox works partially and some parameter needs to be tuned in config params.

which one?

Thank you!

fastchain, 2024-06-07:
Thank you!

> I forgot to ask you: which Pico board do you use?

[image]
[image]

I'll play around with a FF and let you know results.

polhenarejos, 2024-06-07:
Seems a common problem with Firefox.

https://github.com/NixOS/nixpkgs/issues/292134

I tried with FF 126 with macOS and got the same result as you. It works for some people, which makes it more weird. Probably is a problem with permissions.

polhenarejos, 2025-01-15:
No activity. Reopen if needed.

closed 2025-01-15 by polhenarejos

TheMaxMur, 2025-03-23:
I use NixOS unstable (25.05) with rpi pico 2, and get this issue in librewolf (firefox). In chromium key works. Also, my flipper zero have u2f mode, and also works in firefox and chromium. I test this in:
https://webauthn.bin.coffee/
https://webauthn.me/
https://webauthn.io/

also, firefox have about:webauthn for check device information
chromium have chrome://device-log/ for debug

TheMaxMur, 2025-03-23:
Also, i use this module for fido (yubikey)

```nix
{
  config,
  pkgs,
  lib,
  ...
}:
let
  homeDirectory =
    if pkgs.stdenv.isLinux then "/home/maxmur" else "/Users/maxmur";
  
  cfg = config.module.yubikey;
in
{
  options = {
    module.yubikey = {
      enable = lib.mkEnableOption "Enable yubikey support";
    };
  };

  config = lib.mkIf cfg.enable {
    environment.systemPackages = builtins.attrValues {
      inherit (pkgs)
      yubioath-flutter
      yubikey-manager

      pam_u2f
      ;
    };

    services = {
      pcscd.enable = true;
      yubikey-agent.enable = true;

      udev.packages = with pkgs; [ yubikey-personalization libfido2 fuse ];
    };

    security.pam = lib.optionalAttrs pkgs.stdenv.isLinux {
      sshAgentAuth.enable = true;

      u2f = {
        enable = true;

        settings = {
          cue = true;
          authFile = "${homeDirectory}/.config/Yubico/u2f_keys";
        };
      };

      services = {
        login.u2fAuth = true;

        sudo = {
          u2fAuth = true;
          sshAgentAuth = true;
        };
      };
    };
  };
}
```

TheMaxMur, 2025-03-23:
Continue in #129