polhenarejos/pico-fido #47: Non-conformant implementation of supported algorithm search

seesturm, 2024-06-29:
I'm trying webauthn with EdDSA. I understand that pico-fido does not support this. But when sending a pubKeyCredParams list with both supported (ECDSA) and unsupported EdDSA for makeCred operation then pico-fido returns error code CTAP2_ERR_CBOR_UNEXPECTED_TYPE. In my opiniion this not a conformant implementation. [6.1.2. authenticatorMakeCredential Algorithm](https://fidoalliance.org/specs/fido-v2.1-ps-20210615/fido-client-to-authenticator-protocol-v2.1-ps-errata-20220621.html#sctn-makeCred-authnr-alg) states

> If the values of any known members have the wrong type then return an error, for example CTAP2_ERR_CBOR_UNEXPECTED_TYPE.

My interpretation is that the type of the CBOR values, not the values itself are meant. E.g. that an integer was expected but a string was provided. Also there is the Note

> This loop chooses the first occurrence of an algorithm identifier supported by this authenticator but always iterates over every element of [pubKeyCredParams](https://fidoalliance.org/specs/fido-v2.1-ps-20210615/fido-client-to-authenticator-protocol-v2.1-ps-errata-20220621.html#makecred-pubkeycredparams) to validate them.

This further supports my interpretation, that the loop is meant for choosing a supported algorithm, not for checking that all listed algorithms are supported. Can the implementation be changed to simply ignore the entries with unsupported algorithms?

polhenarejos, 2024-06-29:
Fixed in [54bbc0e](https://github.com/polhenarejos/pico-fido/commit/54bbc0e9ea83510bcdf48e3fc61c21180a1882fe).

seesturm, 2024-06-30:
Thank you. Checked out development branch and now my webauthn example is working.

commit 54bbc0e9 (2024-06-29): Fix return value when bad key type is provided. Fixes #47.
closed 2024-07-20 by polhenarejos