polhenarejos/pico-fido #48: Try another key error

0xDRRB, 2024-07-17:
Probably not really an error, but rather a problem of identification of the key by the website.

The site tested is https://passkeys.guru. With firmware 5.8 and USB ID 1050:0407 (Yubikey 5), a popup appears saying “_Try another security key - The security key is not registered with this website (Close) (Retry)_” (sorry for the french screenshot).

[image]

I tested by changing `aaguid[16]` and using the 128 bits from a real Yubikey, same behavior. I also changed the USB strings and the message disappeared, but the key registration didn't work.

There is also  an identification problem with https://passkeys-demo.appspot.com (Google), reporting:  "_This device does not support passkeys._"

It seems that some sites and WebAuthn implementations filter usable keys. But on what information? IDs, strings, AAGUID... Did I miss something?

polhenarejos, 2024-07-17:
Which browser?

0xDRRB, 2024-07-17:
Google Chrome	119.0.6045.199 on GNU/Linux Debian 12.1 amd64.

polhenarejos, 2024-07-17:
Same bug as #47 

If server sends an unsupported algorithm, Pico Fido sends a bad response code, which stops the register process. It has been fixed in the development branch and it will published as a release soon.

0xDRRB, 2024-07-17:
I modified `src/fido/cbor_make_credential.c` manually and indeed, it works.
Even without changing the AAGUID and the strings. Nice.
Thanks a lot!

polhenarejos, 2024-07-20:
Fixed in v5.10

closed 2024-07-20 by polhenarejos