polhenarejos/pico-fido #49: Pico Fido not working on Windows 10 after update from 5.8 to 5.10

reppad, 2024-07-23:
Hi,

First of all, thank you very much for this project, it is very useful !

After updating my key from version 5.8 to version 5.10, I can no longer use it with Windows 10.
It appears in the device manager as before, 4 USB devices with the expected VID:PID (1 Composite, 2 HID & 1 Smart Card), but the browser (tried with Firefox and Edge) or applications like Yubikey Manager and FIDO2.1 Manager no longer detect it.

I use the base variant `pico_fido_pico-5.10.uf2`
I've tried the update using the original binary and the patched binary (Yubikey 4/5), restoring RP2040 flash between the 2 tests, I get the same behavior in both cases.

Everything seems to work correctly in 5.10 with Ubuntu 24.04 and all my settings are preserved (PIN, 2FA, resident keys...).

I didn't lose anything. I had done a memory dump before updating, so I simply restored my key to version 5.8.

If I can provide any other useful information, please don't hesitate to ask :)

polhenarejos, 2024-07-23:
Can you try with Chrome? Firefox has buggy behaviour. I’ll try to debug with win10 to see what’s happening.

reppad, 2024-07-23:
Sure, I'll try it tonight.
There may be a problem with my environment but I doubt it's a problem with Firefox, as Edge doesn't work either, and neither do Yubikey Manager and FIDO2.1 Manager.

reppad, 2024-07-23:
I tried to conduct some more thorough tests, and the behavior is quite strange.
I tested it on several Windows 10 and 11 computers, and the key works randomly. The chosen browser does not seem to influence the behavior.

Sometimes the browser asks to insert the key as if it were not recognized at all, sometimes the browser loops on the PIN code request and the registration does not succeed, and sometimes it works.
Generally, it ends up working after a certain number of attempts; once it works once, it works every time as long as the key is not unplugged.

It's as if a value is at the limit and the functioning depends on the tolerance of the environment.

In any case, reverting to version 5.8 and everything works perfectly, so I don't think it's a hardware problem with my Pico board.

polhenarejos, 2024-07-24:
I cannot reproduce it. I tested in Windows 10 with Chrome, Firefox and Edge against webauthn.io and worked 3/3. Did you try it with webauthn.io or you use a specific page?

reppad, 2024-07-26:
> I cannot reproduce it. I tested in Windows 10 with Chrome, Firefox and Edge against webauthn.io and worked 3/3. Did you try it with webauthn.io or you use a specific page?

In addition to linux -which works well- I tested most combinations of
- Win 10 / Win 11
- Chrome / Firefox / Edge
- https://demo.yubico.com/webauthn-technical/registration / https://www.token2.com/tools/fido2-demo / https://webauthn.io/

I've also done some new tests
- using a fresh installation (using flash_nuke.uf2 before flashing Pico FIDO) as well as updating a key that already had a PIN and passkeys
- using another RP2040 dev board

Same behavior

> Edit: also tested with Yubikey Manager and Fido 2.1 Manager. Did you patch VID/PID with Yubikey values? Mines are 1050:0407

My usual setup is patched as a Yubikey 4/5, I tested with and without.

[image]

I noticed that when the key was ‘not recognised’, there was no need to try several times, just wait between 30 seconds and 1 minute and the ‘Insert your security key’ message disappears and the key is recognised, which is why I found the behavior erratic

Finally, a more accurate description of what I observe is :
Although it appears immediately in the device manager, I have to wait between 30 seconds and 1 minute after connection for the key to be usable on Windows.

rojvv, 2024-08-18:
Same issue. The led doesn't even blink on 5.10 while it does on 5.8. This is regardless of the operating system apparently, since I've tried on both macOS and Fedora.

rojvv, 2024-08-19:
I tried on another Pico. 5.10 makes the led blink, but it does not function at all.

polhenarejos, 2024-08-19:
I spotted several critical problems mixing CCID and HID interfaces. I am preparing a new release, which also adds a support for ESP32.
They are available in the `main` branch.

rojvv, 2024-08-19:
Happy to hear that!

rojvv, 2024-08-21:
I cannot reproduce this issue with Waveshare Zero. I think I will be using that for now.

reppad, 2024-08-26:
> I spotted several critical problems mixing CCID and HID interfaces. I am preparing a new release, which also adds a support for ESP32. They are available in the `main` branch.

I just tried with the HEAD of the `main` branch (`02556fc`).
The key is usable for authentication right from the connection on Windows, so this issue seems to be resolved.
However, the functionality with YubiKey Manager and Yubico Authenticator seems to be broken. I can't manage the active interfaces or access the OTP functions (I'm patched with `1050:0407`).

reppad, 2024-09-02:
> v5.12 should fix all these problems.

Version 5.12 seems to be working well on Windows for authentication.
I don't quite understand why, but it's now necessary to run "Yubico Authenticator" as administrator for it to access the key, which wasn't the case in v5.8.

polhenarejos, 2024-09-02:
What do you mean? You can use Pico Fido with your browser without any external tool.

reppad, 2024-09-02:
Yes, absolutely, and that usage works well.

Yubico Manager and Yubico Authenticator allow you to configure the key, for example the action of the button when the key is not expecting user presence.
It's not a big issue, it's just that it worked fine in v5.8.

polhenarejos, 2024-09-02:
It should work with both, but before you must patch VIDPID with a know Yubico VIDPID in order to those apps recognize your key as a Yubico one.

reppad, 2024-09-02:
Yes, that's what I do in both cases, using Pico Patcher and selecting Yubikey 4/5 OTP+FIDO+CCID.

polhenarejos, 2024-09-02:
What `lsusb -vv` reports?

reppad, 2024-09-03:
v5.12
```
Bus 001 Device 029: ID 1050:0407 Yubico.com Yubikey 4/5 OTP+U2F+CCID
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1050 Yubico.com
  idProduct          0x0407 Yubikey 4/5 OTP+U2F+CCID
  bcdDevice            6.00
  iManufacturer           1 Pol Henarejos
  iProduct                2 Pico Key
  iSerial                 3 DE693C65CB344C36
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0096
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          4 Pico Key Config
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              5 Pico Key HID Interface
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      34
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x04  EP 4 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval              10
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval              10
    Interface Descriptor:
[... 240 lines cut ...]
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
Device Status:     0x0000
  (Bus Powered)
```

reppad, 2024-09-05:
The behavior seems consistent across version regardless of installation method.

Works like real Yubikey, without administrator privileges, except for managing passkeys :
- Version 5.8 installed on a blank Pico
- Version 5.8 installed over version 5.12 (downgrade)

Requires administrator privileges 
- Version 5.12 installed on a blank Pico
- Version 5.12 installed over version 5.8 (upgrade)

polhenarejos, 2024-09-12:
I tried v5.8 and it requires to be ran with Admin privileges, like in v5.12. I do not see any difference in that sense. Perhaps, your OS has whitelisted your previous device.

benallard, 2024-09-24:
I'm having some kind of similar issues with 5.12, Win10 and Firefox.

Pressing reset or removing the key while attempting to generate (register) a key prompt for the device to be re-inserted, but nothing happen after reinsertion. Else the dialog just says please insert your device. The same device works on MacOS.

polhenarejos, 2024-09-24:
I made some adjustments for Windows in 2011cfd3 but the overall behavior seems buggy. Whilst it works smoothly with Edge, in Chrome is not even recognized. I'm not sure if it's a problem of permissions or what.

benallard, 2024-09-24:
Let me know if you need some logs or such, I can try to gather some data there. Should I rather open a separate ticket ?

polhenarejos, 2024-09-24:
First try it with Edge to see whether is a browser specific problem or not.

benallard, 2024-09-24:
will try tomorrow first thing in the morning and let you know 👍

tgrandguillotte, 2024-09-25:
> I'm having some kind of similar issues with 5.12, Win10 and Firefox.
>
> Pressing reset or removing the key while attempting to generate (register) a key prompt for the device to be re-inserted, but [...]

What happens if you leave the pico plugged in for 1 minute before registering (instead of inserting it at the time of the request)?

reppad, 2024-09-25:
> I'm having some kind of similar issues with 5.12, Win10 and Firefox.
>
> Pressing reset or removing the key while attempting to generate (register) a key prompt for the device to be re-inserted, but [...]

What happens if you leave the pico plugged in for 1 minute before registering (instead of inserting it at the time of the request)?

polhenarejos, 2024-09-25:
> So edge is not working either, and only proposing me a QR-code to scan on my phone. as if it would not see the device at all.
> 
> Okay, seems like edge is not working at all with hardware keys as my yubikey is not recognized there either. So I just checked again on firefox, and it's working there, the pico, unfortunately not.

You have to select "use external security key" or similar, the first option. If you get the QR probably you are selecting the second option of "using security key" (not the external security key).

polhenarejos, 2024-09-25:
I made a nightly automatic snapshot to build the firmwares in the development branch. 

https://github.com/polhenarejos/pico-fido/releases/tag/nightly-development

Can you try if it is fixed with latest development firmware?

benallard, 2024-09-25:
It worked!, thanks.
And thanks for modifying the led, that way I was sure I flashed the last version.

metabo7000, 2024-09-28:
> I made a nightly automatic snapshot to build the firmwares in the development branch.
> 
> https://github.com/polhenarejos/pico-fido/releases/tag/nightly-development
> 
> Can you try if it is fixed with latest development firmware?

Developer tell me! v5.8 does not work under windows Acccount (smartcard) mode only under android and linux under Yubico Authenticato under what is the reason?
v5.12 Yubico Authenticato Acccount (smartcard) mode now doesn't even work under linux, it just flashes blue!
[image]

polhenarejos, 2024-09-28:
This is a bug of Yubico Authenticator which its regex only accepts minor versions up to 9. This will be fixed with next version 6.0

metabo7000, 2024-09-28:
> This is a bug of Yubico Authenticator which its regex only accepts minor versions up to 9. This will be fixed with next version 6.0
> 
> This only affects to Smartcard and not ctap/fido.

[image]
But we use the best in publishing!

metabo7000, 2024-09-28:
> This is a bug of Yubico Authenticator which its regex only accepts minor versions up to 9. This will be fixed with next version 6.0
> 
> This only affects to Smartcard and not ctap/fido.

v5.8
this is the question!

linux account is working! :)
[image]

windows account he doesn't work? :(

[image]

v5.12 fault linux and windows os to  account !

FSCSoft, 2024-11-12:
Yo tengo el mimo problema aún ya instale la versión 6.0

closed 2025-01-15 by polhenarejos