polhenarejos/pico-fido #51: How to run on ESP32 boards?

artesonic, 2024-08-04:
I noticed that you said that"Added support for ESP32 boards" in the v5.10 version of the page.But I can't find any relevant information, such as how to compile to the corresponding package, the compiled binary package, etc.What should I do if I want to run this project on my ESP32.
Thanks a lot.

polhenarejos, 2024-08-04:
The ESP32 support was been added to the SDK. At this moment, Pico HSM has full compatibility to ESP32-S3. Pico Fido is under test and will be released soon. The process to flash the ESP device will be pretty similar to Pico HSM: just flash through the web without requiring any external toolchain or building.

artesonic, 2024-08-05:
Thank you very much for answering my question. With your suggestion, I noticed the existence of  [Pico HSM](https://github.com/polhenarejos/pico-hsm) .And I will try to run these two projects on my ESP32.

polhenarejos, 2024-09-03:
ESP32 support has been added in version 5.12.

closed 2024-09-03 by polhenarejos