polhenarejos/pico-fido #52: eddsa for 5.10?

lifetyper, 2024-08-08:
Hi,
Glad to see that version 5.10 is released.
I used to use [Version 5.8 Eddsa 1](https://github.com/polhenarejos/pico-fido/releases/tag/v5.8-eddsa1) for my pico key since I used ed25519 for SSH,.
Can you please release a eddsa supported 5.10 branch?

Thank you.

polhenarejos, 2024-08-20:
I'll make a release for the 5.12 soon. If you need, I've updated the `eddsa` branch so you can test it.

meltingfuse, 2024-08-20:
> I'll make a release for the 5.12 soon. If you need, I've updated the `eddsa` branch so you can test it.

Will eddsa be merged into 5.12 version?If so,I can wait for the 5.12 version release.

polhenarejos, 2024-08-20:
Yes, it is available right now in the repo and a release will be published soon.

closed 2024-09-03 by meltingfuse