polhenarejos/pico-fido #53: Is it necessary to save the master key?

lijicheng168, 2024-08-09:
```
Pico FIDO is an open platform, so exercise caution. The flash memory contents can be easily dumped, potentially revealing private/master keys. It is not feasible to encrypt the content, meaning at least one key (the master key) must be stored in clear text.

If the Pico is stolen, the private and secret keys can be accessed.
```
I think if the master key is not saved, and then each time it is used, the input key is used to decrypt a "master key" into the memory through a symmetric encryption algorithm such as AES, can security be guaranteed?
Thanks.

polhenarejos, 2024-08-09:
This is how Pico HSM or Pico OpenPGP but FIDO works differently not allowing to input a PIN. The PIN requested by FIDO is not transmitted at any time; instead a hash is provided to the device which is then compared with the one previously stored. 

Due to that, I circumvent it by allowing the possibility to transfer the security root to the host (a computer only), which stores a private key used to decipher the master key (we call it “unlock” the master key). However it has important drawbacks:
- It has to be unlocked at every device boot through a Python script. 
- It is tied to an specific host permanently, as it keeps the secret key. It implies the Pico device is not portable anymore since it has to be unlocked by the host at every boot. 
- It breaks the concept of FIDO device, which is the ultimate custodian of your private keys to do not keep it at your computer. If we keep the input key at your computer, we are transfering the trust to the computer again. Of course, it is better than having only the computer, since it would be approached as a “2FA” (pico and host authentication) but it is definitively not the desires approach. 

When you say “input key”, how and when is it transfered to the Pico device?

lijicheng168, 2024-08-15:
Thanks for your explanation.
```
When you say “input key”, how and when is it transfered to the Pico device?
```
I make a double check that it's windows will make a PIN to protect something. Not the pico. You still can 'stealing' private keys from Pico. Is there a way that save the private key in encrypt text and you should decrypt it by some 'PIN' then write it to mem for use. As for how to  “input key”, maybe use the gpios (even little io but with long length) or transfer it by the hid?
Maybe this can help solve the probelm?

polhenarejos, 2024-08-15:
Sure, you can even plug a fingerprint. But the purpose of Picokeys is having just a simple board. The code is open so anyone can adapt it to support 3rd party devices. 

RP2350 seems a good choice as it can store secret keys OTP. It has been released few days ago and I am seeking how to do it. I’ll receive my RP2350 in a couple of days to start playing.

closed 2024-08-19 by lijicheng168