polhenarejos/pico-fido #54: pamu2fcfg -> error: fido_cred_verify (-7) FIDO_ERR_INVALID_ARGUMENT

dietriclX, 2024-08-19:
_I would like to share this error with those of you, who run into the same issue.
After "fixing" this error, I am now able to login without a password ... using the Pico Fido - Pico Keys. A big thanks to those people which made/make this possible._

Even though the pamu2fcfg is working fine with a YubiKey, with the Pico Fido2 it might be failing. At least, that's what I had observed on my on [Debian 12 (bookworm)](https://packages.debian.org/bookworm/pamu2fcfg) system.

```console
$ pamu2fcfg --version
pam_u2f 1.1.0
$ pamu2fcfg
Enter PIN for /dev/hidraw6: 
error: fido_cred_verify (-7) FIDO_ERR_INVALID_ARGUMENT
```

The [version from SID](https://packages.debian.org/sid/pamu2fcfg) is the newest of pamu2fcfg, however comes with two additional new dependencies. So ... I build my own version and in result had been able to retrieve the data required to configure the pam-fido2 process. To be clear, I was only using pamu2fcfg (Version 1.3.0) to retrieving the data from the token.

```console
$ git clone --depth 1 --shallow-submodules --recurse-submodules --branch pam_u2f-1.3.0 https://github.com/Yubico/pam-u2f.git
$ mv pam-u2f pam-u2f.130
$ cd pam-u2f.130
$ sudo apt install --no-install-recommends autoconf automake libtool pkg-config libfido2-dev libpam-dev libssl-dev asciidoc xsltproc libxml2-utils docbook-xml
$ autoreconf --install
$ ./configure
$ make
$ ./pamu2fcfg/pamu2fcfg --version
pamu2fcfg 1.3.0
$ ./pamu2fcfg/pamu2fcfg 
Enter PIN for /dev/hidraw6: 
<UserID>:<KeyHandle1>,<UserKey1>,es256,+presence
`````

polhenarejos, 2024-08-20:
Is the second snippet using Pico Fido or Yubikey?

dietriclX, 2024-08-21:
> Is the second snippet using Pico Fido or Yubikey?

The snippets - I shared in the posting - are done using the Pico Fido.

ihavetenfingers, 2024-08-22:
Cool, this fixed the same issue for me. Thanks!

polhenarejos, 2025-01-31:
Might be related with #91. Try latest nightly development build.

dietriclX, 2025-01-31:
> Might be related with [#91](https://github.com/polhenarejos/pico-fido/issues/91). Try latest nightly development build.

I had given it a try and build it from scratch using Pico SDK 2.1.0 .

Hopefully I took the latest version (nightly source?), by using
```
git clone https://github.com/polhenarejos/pico-fido
git submodule update --init --recursive
:
```
Situation is unchanged ... still getting the same error ... never mind ...

PS: Where do I find these?

polhenarejos, 2025-02-01:
I see the problem.

The thing is that pamu2f does not send the PIN to the authenticator and thus, when you try to log in and get the assertion, it is rejected due to the missing PIN.

For instance, if you try with a fresh firmware without a set PIN, it will work. No PIN is sent, but no PIN is set, so it works. But when you use a PIN, the keys are encrypted with a hash of that PIN. Therefore, you must provide it beforehand to be able to log in. 

The solution is to add `pinverification=1` to your /etc/pam.d/ file. Do not forget to `pamu2cfg` again.

dietriclX, 2025-02-02:
I do have added the "pinverification=1", but the situation is unchanged.

Tested with the firmware from 2 days back.

Pico with freshly installed firmware
```
$ pamu2fcfg
error: fido_cred_verify (-7) FIDO_ERR_INVALID_ARGUMENT
```

Same Pico, after PIN set (`fido2-token -S /dev/hidraw?`)
```
$ pamu2fcfg
Enter PIN for /dev/hidraw4: 
error: fido_cred_verify (-7) FIDO_ERR_INVALID_ARGUMENT
```

=== `/etc/pam.d/common-aut` ===
```
:
# pam-auth-update(8) for details.

auth sufficient pam_u2f.so authfile=/etc/fido2/u2f.keys pinverification=1 cue [c
ue_prompt=Touch Me]

# here are the per-package modules (the "Primary" block)
:
```

FSCSoft, 2026-02-20:
y  tuve ese fallo es el tiempo de presionar del botón esta en 0 y no da tiempo de pisarlo 

[image]