polhenarejos/pico-fido #55: Adafruit QT Py RP2040's led doesnt work

LegitCamper, 2024-08-19:
The QtPy uses a neopixel and has no other leds. I also just flashed a ItysBitsy and it has both a normal led and neopixel. The led works as expected. 

It looks like the neopixel logic is already written but is only enabled on esp boards.

polhenarejos, 2024-08-19:
QtPy and itsy use WS2812, at least from what is defined in `adafruit_qtpy_rp2040.h` and `adafruit_itsybitsy_rp2040.h`. It defines the PIN number 12 and 17 , respectively.

It should be detected automatically on boot. Are you using the firmware [pico_fido_adafruit_qtpy_rp2040-5.10.uf2](https://github.com/polhenarejos/pico-fido/releases/download/v5.10/pico_fido_adafruit_qtpy_rp2040-5.10.uf2) ? Or the equivalent to Itsy. If so, check the PIN number is the proper one.

LegitCamper, 2024-08-20:
Thanks, I did not realize WS2812 was similar to Neopixel. The product page said ''Built-in RGB NeoPixel LED". 

I flashed the [pico_fido_adafruit_qtpy_rp2040-5.10.uf2](https://github.com/polhenarejos/pico-fido/releases/download/v5.10/pico_fido_adafruit_qtpy_rp2040-5.10.uf2), and I also build my own with `-DPICO_BOARD=adafruit_qtpy_rp2040` but the result was the same. 

I also found the pinout sheet, and it also shows 12. 
[image]
Is there another reason my led might not work?

polhenarejos, 2024-08-20:
Hard to say without having one. I have a waveshare rp2040-zero, which also uses the WS2812 and it works. 

If you are skilled, I'd try the example from pico-examples repo: https://github.com/raspberrypi/pico-examples/tree/master/pio/ws2812

My code is based on it, so if the example works, it means something is broken. By the moment, I see the definition of PIO struct has been slightly modified, probably for RP2350. 

Can you build and try the example?

LegitCamper, 2024-08-20:
Well, I built the example and changed the pin to 12 and flashed it and the led still does nothing. I repeated the same for the ItsyBitsy and changed the pin to 17 and got the same result. I must be doing something wrong. It seems odd that I would have two boards with the same behavior.

closed 2024-08-21 by LegitCamper

polhenarejos, 2024-09-03:
Please if you make any progress with this, share it with us. Thanks!