polhenarejos/pico-fido #56: www.picokeys.com is down

rojvv, 2024-08-21:


polhenarejos, 2024-08-21:
I do not observe any problem.

rojvv, 2024-08-21:
This is what Cloudflare says.

[image]

rojvv, 2024-08-21:
And sometimes it just never loads.

[image]

Connection times out.

[image]

rojvv, 2024-08-21:
And here is how making a request with curl looks like.

[image]

rojvv, 2024-08-21:
Everything is alright now.

closed 2024-08-21 by rojvv