polhenarejos/pico-fido #57: change neopixel behaviour?

air-eat, 2024-08-22:
i use this on a waveshare rp2040-one, and it works great (on 5.8; 5.10 seems to be broken on every application and os i tried) but currently the onboard neopixel is configured to flash red at max brightness which is somewhat annoying (to me at least)

maybe, for boards that have a neopixel, configure it to use different colours and brightness for the status? (dim green for active, flashing bright orange for intent check, etc.)

polhenarejos, 2024-09-25:
It shouldn't be changed, I added support for colors, but nothing about the brightness. In idle mode it should blink slowly. Doesn't it happen?

benallard, 2024-09-25:
That's what I meant, the color changed. I could try to investigate how to make it steady if you like.

air-eat, 2024-11-10:
the new colours and brightness control is exactly what i wanted, thanks! i'll close the issue

closed 2024-11-10 by air-eat