polhenarejos/pico-fido #58: pico-fido full flashing and key programming tutorial video windows environment!!

metabo7000, 2024-08-31:
pico-fido full flashing and key programming tutorial video windows environment!!
Unfortunately, installing python v3 modules in a windows environment is a big problem. There are modules that get completely stuck! I'm not a programmer, but a tutorial video would be useful for those who are weak in the python environment!

metabo7000, 2024-08-31:
[image]
problems!
failt probe:(
[https://jhooq.com/pip-install-connection-error/](url)

polhenarejos, 2024-08-31:
You do not need python to flash your board. Just put it in bootsel mode, drag and drop the firmware and it will reboot with your key ready to use.

metabo7000, 2024-09-01:
> Nincs szükség pythonra a tábla flasheléséhez. Csak helyezze bootsel módba, húzza át a firmware-t, és újraindul, amikor a kulcs használatra kész.

Privacy key, how do I add what to use? I want it for a 2fa authentication!

metabo7000, 2024-09-01:
https://www.picokeys.com/2022/07/12/smart-card-shell-3-tool-scs3/

[image]
[image]
GPError: Card (CARD_CONNECT_FAILED/0) - "No card in reader or mute card."
my folder c:\scs3\

polhenarejos, 2024-09-01:
You are mixing apps. Scs3 is intended for Pico HSM, not Pico Fido. To test your pico-fido key as a 2FA you can use webauthn.io

metabo7000, 2024-09-01:
> webauthn.io

would have tried pico_hsm_pico-3.6 firmware, but I only got errors in the pictures above!
But why the hell?

polhenarejos, 2024-09-01:
Pico HSM firmware is not Pico Fido firmware. Please don't be rude.

metabo7000, 2024-09-03:
Esp32-s3 firmware fido decompiler?
vscode modul name?

Bjelko, 2024-09-08:
> Esp32-s3 firmware fido decompiler? vscode modul name?

why do you need decompiler if you have source code?

metabo7000, 2024-09-08:
> > Esp32-s3 firmware fido decompiler? vscode modul neve?
> 
> miért kell decompiler ha van forráskódod?

compile source code to firmware file bin extension! How and in what environment do you do this?
I tried pico sdk under vscode but totally picora!
Espr32 not even that much!
Could you help me get started, what should I do?

metabo7000, 2024-09-08:
compiler problem?
[image]
[image]