polhenarejos/pico-fido #59: failed to load resident ssh key,Err code:Unable to load resident keys: device not found

meltingfuse, 2024-09-06:
Hi
I'm running 5.12 eddsa on my pico board,using openssh windows from[ this repo](https://github.com/PowerShell/Win32-OpenSSH/releases).
I managed to created resident ssh key and save it to the pico key,but it never allow me to readout the resident key using `ssh-add -K -S internal` or `ssh-keygen -K`.
Using these command under a admin mode powershell env give me a result like this
```
ssh-keygen -K -v
Enter PIN for authenticator:
You may need to touch your authenticator to authorize key download.
debug1: find_helper: using "C:\\Program Files\\OpenSSH\\ssh-sk-helper.exe" as helper
debug1: client_converse: helper returned error -60
Unable to load resident keys: device not found
```
I can see that the led light turned to Press to confirm mode,but after I pressed the button,it turned to idle mode rather than processing mode.
Also there is something wrong with the hid device driver after I pluged in the pico key.
[image]

Everything works fine when I use the key to login website with webauthn.

It even works when I login ssh with -i option to load the key file from local hard disk like:
`ssh root@myserver -v -i .\id_ed25519_sk`

I just can't use the ssh-add or ssh-keygen command to  dump the key file from pico key to local disk.

metabo7000, 2024-09-08:
> Hi I'm running 5.12 eddsa (with vendor patch to Yubikey 4/5 OTP+FIDO+CCID )on my pico board,using openssh windows from[ this repo](https://github.com/PowerShell/Win32-OpenSSH/releases). I managed to created resident ssh key and save it to the pico key(pretty sure about it since I see the light in processing mode and If I create a new ssh key with the same name it tells me there is already a key with same name there).
> 
> But it never allow me to readout the resident key using `ssh-add -K -S internal` or `ssh-keygen -K`. Using these command under a admin mode powershell(has to run under admin mode since this is a bug for openssh windows version)give me a result like this
> 
> ```
> ssh-keygen -K -v
> Enter PIN for authenticator:
> You may need to touch your authenticator to authorize key download.
> debug1: find_helper: using "C:\\Program Files\\OpenSSH\\ssh-sk-helper.exe" as helper
> debug1: client_converse: helper returned error -60
> Unable to load resident keys: device not found
> ```
> 
> I can see that the led light turned to press to confirm mode,and after I pressed the button,it turned to active mode rather than processing mode. Also there is something wrong with the hid device driver after I pluged in the pico key. ![Screenshot 2024-09-07 000959](https://private-user-images.githubusercontent.com/177820002/365235096-9c251fa0-a5af-40d8-9ce9-29223ad644e1.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjU3ODg4NzUsIm5iZiI6MTcyNTc4ODU3NSwicGF0aCI6Ii8xNzc4MjAwMDIvMzY1MjM1MDk2LTljMjUxZmEwLWE1YWYtNDBkOC05Y2U5LTI5MjIzYWQ2NDRlMS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTA4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkwOFQwOTQyNTVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0yMTViMGY0ZjNmMmYyMzhhM2ZmZjhhZDRlZDEzMGRlNzg4MDliNTQ4Y2UwMWI0ZTg5MDcyZDc4YWUxNzE4MTBjJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.bjAbEpNnrqZfZW_YOIrRobgGhKmcyUqI4lC2lWtQ_3Y)
> 
> But Everything else works fine like using the key to login website with webauthn.
> 
> It even works when I login ssh with -i option to load the key file from local hard disk like: `ssh root@myserver -v -i .\id_ed25519_sk`
> 
> I just can't use the ssh-add or ssh-keygen command to dump the key file from pico key to local disk.
> 
> I've tried several board like official pico board, waveshare pico zero and some board I made by myself,got the same result. Could it be the driver issue or something else?
> 
> Update: I just did some more test and confirmed that this issue(ssh key load and yellow exclamation mark in Device Manager) happens since 5.10 version,rollback to 5.8 solves the problem.But 5.8 has some other issues fixed in 5.10 and 5.12 which bother me more.
> 
> On the same pc(windows 11 os),using **5.8 version everything seems fine in the device manager**,
> 
> with 5.10 version: ![5 10_error](https://private-user-images.githubusercontent.com/177820002/365327001-0762ed60-decc-4d0d-bb0c-5edb3fa17764.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjU3ODg4NzUsIm5iZiI6MTcyNTc4ODU3NSwicGF0aCI6Ii8xNzc4MjAwMDIvMzY1MzI3MDAxLTA3NjJlZDYwLWRlY2MtNGQwZC1iYjBjLTVlZGIzZmExNzc2NC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTA4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkwOFQwOTQyNTVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT04OWVlMDEzM2EwYmRhM2IxNWQxNjVlMGVhNmY0YmZjZTliMzMyNzk2OWY5MDAwODYzZjc2Y2RlOTY5ZWQ4Njc1JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.cPvR-u16QgmeyyapsWHn6crIpGxsNuk6rJys3yMn3X4)
> 
> with 5.12 version: ![5 12_error](https://private-user-images.githubusercontent.com/177820002/365327073-324ae4ba-2725-4017-ab55-b924c48cbdfa.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjU3ODg4NzUsIm5iZiI6MTcyNTc4ODU3NSwicGF0aCI6Ii8xNzc4MjAwMDIvMzY1MzI3MDczLTMyNGFlNGJhLTI3MjUtNDAxNy1hYjU1LWI5MjRjNDhjYmRmYS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTA4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkwOFQwOTQyNTVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT00OGNiYzM0OWM0NGRlMmI2OTBkNWY2M2M1M2M5OTcyMzhiNzg4MWI4YTVhZDNlMjY5ZWUwODM3YzQ4NDZmODQyJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.9rJKZ9PZz43U_UAEIqo0eGvyGJIuBZMS87UD5-NzZkM)
> 
> No difference between eddsa branch or main branch.

yubico 5 and neo fido_5.8 firmware works great as it should, Yubico Authenticator and YubiKey Personalization Tool handle it, but fido 5.10 and 5.12 are completely defective, yubico manager programs don't handle them completely error-free either! The developer gives very few answers here on the forum, documentation is incomplete, hms versions are still we will never be able to communicate under Windows/Linux or with the recommended environmental software! tutorial video would help a lot for hsm device!

polhenarejos, 2024-09-13:
> I managed to created resident ssh key and save it to the pico key(pretty sure about it since I see the light in processing mode and If I create a new ssh key with the same name it tells me there is already a key with same name there).

How are you sure it is in the device? How do you generate it?

polhenarejos, 2024-09-13:
Fixed in ec612a4.

meltingfuse, 2024-09-13:
> > I managed to created resident ssh key and save it to the pico key(pretty sure about it since I see the light in processing mode and If I create a new ssh key with the same name it tells me there is already a key with same name there).
> 
> How are you sure it is in the device? How do you generate it?

I use command as below to generate the key
`ssh-keygen -t ed25519-sk -O resident -O application=ssh:YourTextHere -O verify-required`

I'm sure it is in the device since I generated it with a pico key with 5.8 version and everything works fine,just upgrading a firmware version won't erase the key already exist,right?

Thank you for the fix,I will try to build and verify it later.

meltingfuse, 2024-09-14:
> Fixed in [ec612a4](https://github.com/polhenarejos/pico-fido/commit/ec612a451da5e8cbee61de866dd273f214e6cd5c).

H,I just compiled the development branch and confirmed that this commit,can you please merge it into the eddsa branch?

Also I found a problem that I have to use flash_nuke.uf2 file to reset the pico board（which erase everything on the spi flash I think） then reinstall the new firmware.If I just put it into boot mode and upgrade with the compiled firmware the problem still exist.

HeavyDutySoul, 2024-09-15:
I try on 5.12-eddsa1 branch with ssh-fix. With no luck. :(
`ssh-keygen -t ed25519-sk -Ono-touch-required -Oresident -Overify-required -Oapplication=ssh:picokey -Odevice="\\\\?\\hid#vid_1050&pid_0407&mi_00#7&254315e3&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}" -N '' -C "ssh:github.com" -v`
Generating public/private ed25519-sk key pair.
You may need to touch your authenticator to authorize key generation.
debug1: start_helper: starting /usr/lib/ssh/ssh-sk-helper 
debug1: sshsk_enroll: provider "internal", device "\\\\?\\hid#vid_1050&pid_0407&mi_00#7&254315e3&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}", application "ssh:picokey", userid "(null)", flags 0x24, challenge len 0
debug1: sshsk_enroll: using random challenge
debug1: check_enroll_options: requested device \\\\?\\hid#vid_1050&pid_0407&mi_00#7&254315e3&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
debug1: ssh_sk_enroll: using device \\\\?\\hid#vid_1050&pid_0407&mi_00#7&254315e3&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
debug1: check_sk_options: option uv is unknown
debug1: key_lookup: fido_dev_get_assert: FIDO_ERR_RX
debug1: ssh_sk_enroll: key_lookup failed
debug1: sshsk_enroll: provider "internal" failure -1
debug1: ssh-sk-helper: Enrollment failed: invalid format
debug1: main: reply len 8
debug1: client_converse: helper returned error -4
Key enrollment failed: invalid format

`ssh-keygen -K -Odevice="\\\\?\\hid#vid_1050&pid_0407&mi_00#7&254315e3&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}"`
Enter PIN for authenticator: 
You may need to touch your authenticator to authorize key download.
Provider "internal" returned failure -1
Unable to load resident keys: invalid format

`ssh -V`
OpenSSH_9.8p1, OpenSSL 3.2.1 30 Jan 2024

meltingfuse, 2024-11-05:
> Fixed in [ec612a4](https://github.com/polhenarejos/pico-fido/commit/ec612a451da5e8cbee61de866dd273f214e6cd5c).

Can you please merge this commit to the eddsa branch?Thank you.

meltingfuse, 2024-11-08:
Sorry I'm reopening this issue since eddsa support is really import for many ppl.

commit ec612a45 (2024-09-13): Fix ssh-keygen creation.

Fixes #59
closed 2024-11-08 by meltingfuse