polhenarejos/pico-fido #60: esp32-s3 bin

metabo7000, 2024-09-08:
Why is there no esp32s3 binary version available?
we need pico version 5.8!
We cannot do source code translation!
Tankyou!

metabo7000, 2024-09-14:
Unfortunately, I still haven't managed to compile the source under vscode, neither under windows nor under linux, I get a cmake error, I do this, pico sdk and gcc engines are installed and activated!
I found these, but I really need esp32 pico version 5.8, it's stable under rp2040!
Why is there no information on how to solve this?

https://raw.githubusercontent.com/polhenarejos/polhenarejos.github.io/main/esp/fw/pico_hsm_esp32-s3_4.2.bin
https://raw.githubusercontent.com/polhenarejos/polhenarejos.github.io/main/esp/fw/pico_fido_esp32-s3_5.12.bin

metabo7000, 2024-09-14:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
Is everyone going their separate ways?

polhenarejos, 2024-09-14:
ESP32 support came in v5.12. Earlier versions did not provide it yet.

metabo7000, 2024-09-14:
> ESP32 support came in v5.12. Earlier versions did not provide it yet.

pico_fido_esp32-s3_5.12.bin edit vid/pid rename?
[image]

polhenarejos, 2024-09-14:
It’s not supoorted yet for ESP32. You must pass USB_VID and USB_PID to cmake with your values.

metabo7000, 2024-09-14:
> It’s not supoorted yet for ESP32. You must pass USB_VID and USB_PID to cmake with your values.

Where do you have to get it?
In your source?
for example 1050:0407!

metabo7000, 2024-09-14:
> It’s not supoorted yet for ESP32. You must pass USB_VID and USB_PID to cmake with your values.

They can't even translate the source code, how do I make a set esp32 firmware!
Why don't you prepare and upload gibhub 1050/0407 here?
You are only wasting precious time with your secrecy!

polhenarejos, 2024-09-14:
Because it’s a law infringement.

metabo7000, 2024-09-14:
> Because it’s a law infringement.

so can i do hex editor bin file?

closed 2025-01-15 by polhenarejos