polhenarejos/pico-fido #61: No LED on W

peardox, 2024-09-11:
CMakeLists.txt needs something like...

> # Configure compilation flags and libraries# Cater for Pico W
> if (PICO_CYW43_SUPPORTED)
>     message(STATUS "It's a Pico W.....")
>     target_link_libraries(pico_fido PUBLIC pico_cyw43_arch_none hardware_clocks)
> endif()
> 

Otherwise in main() PICO_DEFAULT_LED_PIN is set and CYW43_WL_GPIO_LED_PIN isn't

polhenarejos, 2024-09-11:
Do you pass `PICO_BOARD=pico_w` to CMake?

peardox, 2024-09-11:
Yep - via ...

> set(PICO_BOARD pico_w CACHE STRING "Board type")

I notice your latest Release build also doesn't do the LED

Odd really as pico-keys-sdk/pico_keys_sdk_import.cmake has a test for it and includes correct lib

Then again building on Windows prob doesn't help at my end

peardox, 2024-09-11:
Sorry - just discovered one (of five) Ws I tried pico-fido on is different from the others

Turns out it works fine on four of them but fails on the other (which is the one I tried first)

The blinking idle is very distracting. Got a bunch of Waveshare-RP2040-GEEKs as they have inbuilt 1.14" 7789 LCDs (and were dirt cheap from China). Wanna change blink into LCD telling you what it wants you to do next.

closed 2024-09-11 by peardox