polhenarejos/pico-fido #62: Use Secure Boot on RP2350 based boards like RP Pico 2

java-py-c-cpp-js, 2024-09-13:
AFAIK the secure boot feature could prevent flashing an alternative firmware to extract the keys saved.
For alternative products like Yubikey, Solokey etc. this is an important feature.

See [rp2350-datasheet.pdf](https://datasheets.raspberrypi.com/rp2350/rp2350-datasheet.pdf) page 420 __5.10.1. Secure Boot__ ff.

polhenarejos, 2024-09-13:
Secure Boot will not solve it per se, since you can still dump the contents of flash. `picotool` is able to download and save the entire flash in your disk in a couple of minutes. 

The approach I will follow in the next release is:
1- Enabling secure boot to avoid non-authorized firmware to access the keys.
2- Enable OTP to store master keys.
3- Protect OTP region data from being accessible with external tools like `picotool`.
4- Invalidate other secure boot keys, to avoid an attacker may burn their keys.
5- Disable all debug interfaces for obvious reasons.
6- Protect bootrom from booting older releases to avoid flash exploitable releases, even they are signed with the same bootkey.
7- Run the program in RAM to avoid flash sniffing, with the consequences of such limited space. 

Note that having the flash outside the die is still an attackable vector. This hopefully will be solved in RP2354 but I do not know when it will be available, probably by 2025 or later.

All of these points require to perform modifications onto your board that are not regressive and are permanent. If your purpose is to use your board *only* for Pico Fido, then it is fine. But in case you are just testing or you want to use your board for other purposes, then this level of security it is not suitable for you.

In any case, next release will contain an upgrade on the security including all these aspects, leaving the Secure Boot as optional for anyone. A master key will be burnt in the OTP area, which will protect all your keys and will be unaccessible from outside (unless RP2350 has a 0-day vulnerability). This will protect all your data with a secret AES256 key and this will be a major step by far since RP2040. If you are interested, there is a bunch of updates in the `development` branch that include OTP and secure boot.

polhenarejos, 2024-11-26:
Added in v6.0

closed 2024-11-26 by polhenarejos