polhenarejos/pico-fido #63: Building image for esp32s3 target

halsbox, 2024-09-16:
Hi!
What branch/tag should I use for building esp32s3 images?
What's the command to build for esp32s3? 
What env variables should be set when building image?
Where in source tree (including git submodules) should I search for and add my own board .h files (e.g. to set LED type/pins)
I did a brief grep through source tree (including *-sdk) on master branch and tried running cmake with ESP_PLATFORM=1 but didn't manage to get it building.
Will appreciate any help.

polhenarejos, 2024-09-17:
You don't need nothing special. 

If you have configured your VS Code with ESP32 plugin, the project is built smoothly. 

https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/tutorial/install.md

halsbox, 2024-09-17:
Thanks! Got it building with custom vid/pid.
On linux I had to patch pico-keys-sdk/src/esp_compat.h:
```
diff --git a/src/esp_compat.h b/src/esp_compat.h
index 0379c13..6266058 100644
--- a/src/esp_compat.h
+++ b/src/esp_compat.h
@@ -20,7 +20,7 @@
 
 #ifdef ESP_PLATFORM
 
-#include "freertos/freertos.h"
+#include "freertos/FreeRTOS.h"
 #include "freertos/queue.h"
 typedef QueueHandle_t queue_t;
 #define queue_init(a,b,c) do { *(a) = xQueueCreate(c, b); } while(0)
```
Also found nothing VSCode specific. Any or no IDE is OK, general instructions would be:
```
git clone --recursive https://github.com/espressif/esp-idf.git && cd esp-idf
./install.sh esp32s3
. ./export.sh
cd /path/to/pico-fido/
idf.py -DUSB_VID=0x**** -DUSB_PID=0x**** all
idf.py flash
```

metabo7000, 2024-09-18:
> Thanks! Got it building with custom vid/pid. On linux I had to patch pico-keys-sdk/src/esp_compat.h:
>
> ```
> diff --git a/src/esp_compat.h b/src/esp_compat.h
> index 0379c13..6266058 100644
> --- [...]

Can you show me how you do this?
I would be very grateful for a tutorial video!
Developer does not help!

halsbox, 2024-09-18:
Hi, @metabo7000!
Please provide what OS do you use.
I did build it using my Linux desktop. For linux you need git installed and do:
1. 
```
git clone --recursive https://github.com/polhenarejos/pico-fido.git
```
2. change `#include "freertos/freertos.h"` to `#include "freertos/FreeRTOS.h"` in pico-fido/pico-keys-sdk/src/esp_compat.h
3. Follow "Step 1. Install Prerequisites" from https://docs.espressif.com/projects/esp-idf/en/stable/esp32/get-started/linux-macos-setup.html#get-started-prerequisites
4. (replace asterisks with your desired vid and pid)
```
git clone --recursive https://github.com/espressif/esp-idf.git && cd esp-idf
./install.sh esp32s3
. ./export.sh
cd .../pico-fido/
idf.py -DUSB_VID=0x**** -DUSB_PID=0x**** all
idf.py flash
```

halsbox, 2024-09-18:
@metabo7000 just checked docker build is OK. 
If you own any OS with docker installed you can build using docker. 
1. clone https://github.com/polhenarejos/pico-fido.git including submodules using you preffered tool.
2. patch `pico-fido/pico-keys-sdk/src/esp_compat.h` to include case sensitive "freertos/FreeRTOS.h" as I as I mentioned above 
3. go to cloned directory (e.g. `cd pico-fido`) and run:
docker run --rm -v $PWD:/pico -w /pico -u $UID -e HOME=/tmp espressif/idf idf.py -DUSB_VID=0x**** -DUSB_PID=0x**** all
```
3. your build firmware binaries are located  in `build` subfolder.

metabo7000, 2024-09-20:
> Hi, @metabo7000! Please provide what OS do you use. I did build it using my Linux desktop. For linux you need git installed and do: 1.
>
> ```
> git clone --recursive [...]

[image]
[image]
[image]
[image]
I did as I described! But I get these errors! I only worked in arduino ide environment in window os!

halsbox, 2024-09-21:
@metabo7000 you skipped all steps except for first one and last one, that's why you get all this errors. If you are not familiar with command line and still want to build pico-fido by yourself, I'd recommend using VSCode and following
https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/tutorial/install.md (choose Express setup option).
As soon as you have VSCode with ESP-IDF Extension installed, you should open pico-fido project, select esp32s3 target (2) and click build (3) in ESP-IDF explorer (2):
[image]

metabo7000, 2024-09-21:
> @metabo7000 you skipped all steps except for first one and last one, that's why you get all this errors. If you are not familiar with command line and still want to build pico-fido by yourself, I'd recommend using VSCode and following https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/tutorial/install.md (choose Express setup option). As soon as you have VSCode with ESP-IDF Extension installed, you should open pico-fido project, select esp32s3 target (1) and click build (3) in ESP-IDF explorer (2): ![screncapture](https://private-user-images.githubusercontent.com/7151279/369601883-8d4881c5-0467-49c5-8683-8564472236b4.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjY5MTc4MDEsIm5iZiI6MTcyNjkxNzUwMSwicGF0aCI6Ii83MTUxMjc5LzM2OTYwMTg4My04ZDQ4ODFjNS0wNDY3LTQ5YzUtODY4My04NTY0NDcyMjM2YjQucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDkyMSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA5MjFUMTExODIxWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YzQzMGZmY2RjM2Q1ZDRmNTRmNzY1NDY5NjMyOGFlMWM0NmE4ZDNkN2I2NDk4Yzc4MTI4YzA1NzUyNzZkZTc0NyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.Z8atmVKYIoQthkSfc6g_NffcjwoXY30E7jTdd1uSWwE)
hi!
[image]
linux problémákhoz nem értek soha nem használtam ezt az OS!

metabo7000, 2024-09-21:
[image]
This translation of 5.12 idf 2000 lines is just a mistake!

metabo7000, 2024-09-21:
> @metabo7000 you skipped all steps except for first one and last one, that's why you get all this errors. If you are not familiar with command line and still want to build pico-fido by yourself, I'd [...]

Please make me a fido firmware bin esp32s3 in which 1050/0407 vid/pid settings are ready!
You can send it to metabo7000@gmail.comTankyou! !
The translation cannot solve your problem, see the attached pictures!

metabo7000, 2024-09-21:
idf v5.4 ,v5.3 version compile error!
video
[https://drive.google.com/file/d/17NbH_wvGrV0PUWCVdCHUyRRd1avLqcIu/view?usp=sharing](url)

halsbox, 2024-09-21:
> ![Snap015](https://private-user-images.githubusercontent.com/171061281/369618286-7511b473-28b8-41ee-94cd-2b1a2b519898.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjY5NTg0OTAsIm5iZiI6MTcyNjk1ODE5MCwicGF0aCI6Ii8xNzEwNjEyODEvMzY5NjE4Mjg2LTc1MTFiNDczLTI4YjgtNDFlZS05NGNkLTJiMWEyYjUxOTg5OC5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTIxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkyMVQyMjM2MzBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1mZDBmM2Y0ZmZiMzA1Yjk0NTIyYjg2MzY0OWQ2MDBjYTQzZjM0NGMxZmM0NzI0MjgzNjBhMzNhOGRjOGY3ZDcyJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.vtM3Qa8_oTz0ZjK3v4KmcnzSZlPtuYioNtDlj1a99IQ) This translation of 5.12 idf 2000 lines is just a mistake!

Those are not build errors, that's warnings mostly about style of code, don't pay attention to them, don't try that's not on the provided instruction, just carefully do the steps. Also check your email and please stop spamming issues on this repo. Github issues are not intended for that kind of support you are seeking here. If you have problems with operating VSCode or ISP-IDF you should ask on corresponding resources.

closed 2024-09-21 by halsbox

metabo7000, 2024-09-22:
> > ![Snap015](https://private-user-images.githubusercontent.com/171061281/369618286-7511b473-28b8-41ee-94cd-2b1a2b519898.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjY5NTg0OTAsIm5iZiI6MTcyNjk1ODE5MCwicGF0aCI6Ii8xNzEwNjEyODEvMzY5NjE4Mjg2LTc1MTFiNDczLTI4YjgtNDFlZS05NGNkLTJiMWEyYjUxOTg5OC5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTIxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkyMVQyMjM2MzBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1mZDBmM2Y0ZmZiMzA1Yjk0NTIyYjg2MzY0OWQ2MDBjYTQzZjM0NGMxZmM0NzI0MjgzNjBhMzNhOGRjOGY3ZDcyJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.vtM3Qa8_oTz0ZjK3v4KmcnzSZlPtuYioNtDlj1a99IQ) This translation of 5.12 idf 2000 lines is just a mistake!
> 
> Check your email and please stop spamming issues on this repo. Github issues are not intended for that kind of support you are seeking here. If you have problems with operating VSCode or ISP-IDF you should ask on corresponding resources.

The supported resources are those who help others as you do! There are those who, for example, only understand others, but also open or eat in other areas! Thank you for the filmware files!

commit 6f517e8f (2024-09-19): Fix header in Linux. Fixes #63