polhenarejos/pico-fido #64: pico patcher tool doesn't work with 5.12

dragonnn, 2024-09-24:
If I upload 5.10 file for my waveshare one it works fine, file is downloaded.
When I select 5.12 file and press nothing happens.
I also tried running local the [pico-fido-patch-vidpid.sh](https://github.com/polhenarejos/pico-fido/blob/main/pico-fido-patch-vidpid.sh), it seam like it run fine but when I flashed the generated uf2 to my waveshare one it didn't show up with the right VID:PID I put to that script.
Any idea?

dragonnn, 2024-09-24:
I did mean the web page pico patcher, this one: https://www.picokeys.com/pico-patcher/ doesn't work for me with 5.12. Tried with both Vivaldi and Firefox to rule out a web browser issue (and it works fine with 5.10).

polhenarejos, 2024-09-24:
Which VIDPID do you use?

dragonnn, 2024-09-24:
I tested some custom ones and a few presets (mostly interested in the Yubikey ones). 
[image]
Here is how it looks on my PC, I press PATCH in this state and nothing happens. If I select pico_fido_waveshare_rp2040_one-5.10.uf2 I get a patched file on that same page.

polhenarejos, 2024-09-24:
Interestingly only happens with ` pico_fido_waveshare_rp2040_one-5.12.uf2`

dragonnn, 2024-09-25:
At least it is reproducible :D

polhenarejos, 2024-09-25:
Can you try this release? https://github.com/polhenarejos/pico-fido/releases/tag/nightly-development

dragonnn, 2024-09-25:
That worked, thanks!

closed 2024-09-25 by dragonnn