polhenarejos/pico-fido #65: Does not work with Firefox

tutoduino, 2024-10-04:
Hello,
PicoFIDO fails to run on Raspberry Pi Pico with Firefox 131.0 under Linux Mint. 
Testing on https://www.token2.com/tools/fido2-demo
It works fine with Chromium, but I need this solution to work with Firefox...
Thanks for your support.

polhenarejos, 2024-10-07:
Can you try flashing with a nightly development build?

tutoduino, 2024-10-08:
I did, the result of registration is "operation cancelled" after PIN code is entered and button is pushed.
See attachment.
Thanks for your support.
[image]

polhenarejos, 2024-10-08:
And it works well with Chrome, right?

polhenarejos, 2024-10-09:
With Firefox in Windows 10 works. Not sure if it is a linux problem. However, compared to Chrome, Firefox is extremely slow. It takes almost 1 minute to perform the process, whereas Chrome takes 2-3 seconds.

tutoduino, 2024-10-09:
Interesting thanks, then I tried on Firefox under Windows (I do not have Windows at home, so need to test another system :)) and indeed it works. 
I am exploring the permission under Linux, because when I run the FIDO2 Key Data Explorer in Firefox/Linux I get an user permission error... 
[image]

polhenarejos, 2024-10-09:
I did not succeed in macOS neither. Firefox is sending continuously a bogus AAID for an unknown reason, which hangs the pico fido until it is closed. Perhaps is related. 

If you are skilled, you can build the firmware with “DEBUG_APDU=1” in cmake and debuging UART0. I use a TTL to USB dongle that costs a couple of dollars using a terminal opened to the exposed COM port. I can review the log then.

tutoduino, 2024-10-09:
On my setup (Firefox/Linux), my 2 other "real" FIDO2 key work fine.

tutoduino, 2024-10-10:
Compilation fine, on which pins did you configure UART0 and at which speed ? 
Thanks

polhenarejos, 2024-10-10:
Nothing else?

tutoduino, 2024-10-10:
No

tutoduino, 2024-10-10:
I register the key on Chrome successfuly (PIN is setup). And then I try again to register on Firefox. 
The first steps are fine, I enter the PIN, touch the key, then the message operation cancelled appears. 
Here are the traces in this case. 
[putty.log](https://github.com/user-attachments/files/17333391/putty.log)

polhenarejos, 2024-10-10:
Does it work with https://webauthn.io ?
Did you test it with no cookies and blank pico fido? Withour resident keys. 

Perhaps this helps to debug

https://addons.mozilla.org/en-US/firefox/addon/webdevauthn/

tutoduino, 2024-10-11:
I tested with no cookies and blank pico without resident keys. 
Interesting warning on webauthn.io, looks like permission issue !
I do not understand how to use the webdevauthn addon, where are displayed the debug information ?
[image]

tutoduino, 2024-10-11:
I checked the forums and found https://webauthn.bin.coffee/  testing site that raises similar issue. 

[image]

polhenarejos, 2024-10-11:
Related with #46

polhenarejos, 2024-10-11:
Does it work?

"Adding on to the previous answer by Cody, and the respective comments, you must also enable `security.webauth.u2f` in Firefox's `about:config` for U2F, FIDO2, and subsequently Yubikeys to work."

tutoduino, 2024-10-12:
By default, firefox config set webauth flags correctly (flag names are different now, see attached picture). All flags set to true doe not work better.
[image]

tutoduino, 2024-10-12:
I see on some [forums](https://support.mozilla.org/mk/questions/1412073) that downloading Firefox from the official repository solved their issue.
I tried but it does not work for my problem here...

polhenarejos, 2024-10-12:
And enabling direct attestation?

tutoduino, 2024-10-12:
tried to enable all, does not work better

tutoduino, 2024-10-12:
Firefox support could help ?

polhenarejos, 2024-10-12:
What is strange to me is that your true Yubikey works smoothly with Firefox. There's something in it that is recognized by Firefox somehow.

tutoduino, 2024-10-12:
Yes indeed. An I suspected udev but pico works fine with chromium...

polhenarejos, 2024-10-16:
Any progress?

tutoduino, 2024-10-16:
no idea of ​​research path...

LinOx2, 2024-10-24:
I have the same issue with Firefox.
I use Falkon (chromium) for register, after I can use Firefox (131.0.3) for login, it work like that for me.

Falkon (webkit) on linux. Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Falkon/22.04.3 Chrome.

so Firefox cancel only register. BUGs in Firefox ???.

tutoduino, 2024-10-24:
Firefox works fine with other FIDO2 keys so unclear to me...

LinOx2, 2024-10-25:
I have Linux Mint and the same issue, and I use Raspberry Pi Pico (rp2040) in the way I have explain.
firefox reject (cancel) the pico key at register, but work well at login, if i have register with Falkon the pico key.
Maybe reading the Firefox code on fido register could explain what the problem is.

patvdleer, 2024-11-01:
I can confirm I have the same issues, I'm using the Waveshare RP2040 One running on `Ubuntu 22.04.4 LTS` with `6.8.0-45-generic`.  I have the `Yubico NEO OTP+U2F+CCID` and the `SoloKeys Solo 4.0.0` which do work both on Firefox and Chrome.

Since I bought multiple waveshares I have one with and one without a pin set, both don't work.

Any way to further debug this?

Firefox settings:
[image]

polhenarejos, 2024-11-01:
So which is the combination that does not work?

patvdleer, 2024-11-01:
On Linux/Ubuntu on firefox the picokey doesnt work, others do. The picokey does work on Chrome.

On Mac it does work on Firefox and Chrome.

polhenarejos, 2024-11-01:
I suspect it is related with the CCID interface. Firefox sends APDUs trying to find a specific file reference, but I do not know which one is the good one present in Yubikeys or Solo. 

If you want to debug, you can try to generate a log file of pcscd process. More info at https://ccid.apdu.fr/ (see Log section). Do it with your Yubikey. Plug the Yubikey and start Firefox. 
In the log file we’ll see the communication exchanged with the Yubikey and Firefox and I hope we’ll find which are the references needed by Firefox.

air-eat, 2024-11-10:
i don't know if this is related, but disabling `security.webauthn.ctap2` for me seems to make it pass [various](https://demo.yubico.com/webauthn-technical) [tests](https://webauthn.bin.coffee/), and registering/signing in, although not through the "Or, sign in with passkey option", works with discord (even though none of the examples ask for the pin or recognize the device)

however, it does fail immediately in other areas like github passkey registration

air-eat, 2024-11-10:
btw, i get the same behaviour with firefox on both macos and linux

leecheeyong, 2024-12-15:
I'm not sure if this is related.

So I had everything working well before updating to macOS 18, on Firefox. Previously when the mac prompt the take action on the security key box, the Pico would flash on and off with a delay of 1 second but now when the box prompt, the pico will either keep the led on forever or kept off. It seems like the mac literally doesn't see it security key or something. I thought something was corrupted on my Pico, so I even tried re-flashing the firmware with the latest ones (version 6) but this time when it is plugged in, the pico wouldn't even blink and be detected for some reason. And then I tried going down version by version and ended up in version 3 where it will blink when plugged in but face the same issue as what happened initially.

On Windows however, it would be detected and blink as it should, but sometimes I'll have to reconnect the device multiple times for it to actually work as it would rapidly blink but does nothing sometimes.

polhenarejos, 2024-12-15:
Can you try the nightly development build? There were some issues fixed recently.

leecheeyong, 2024-12-17:
> Can you try the nightly development build? There were some issues fixed recently.

Yep, I've tried all of them, and it seems like its an issue on the mac itself but I don't know what it is

air-eat, 2024-12-18:
yubikey fido test and github login still error out in the same way but now at least ask for a pin and presence auth, so it seems like a step in the right direction

polhenarejos, 2024-12-18:
Can you check no NSS modules are interfering? I had a problem in the past with one PKCS11 module that stole the connection.

You can check it at Settings > Privacy & Security > Certificates > Security Devices 
I have only 3 enabled devices:
- NSS Internal
- Internal root
- OSSClient

air-eat, 2024-12-20:
if i had any, it wouldn't explain the identical behavior across two different devices (pc with linux, laptop with macOS)

polhenarejos, 2024-12-20:
Usually OS handle smartcards differently. macOS relies on libccid, Windows relies on scard driver. And despite pico-fido is mainly used by hid interface, it has a ccid interface that may lock the card if a daemon requires it. I had this problem with firefox in macOS. Whilst it worked in Windows it did not work with macOS, due to the PKCS11 module in firefox.

randalthor17, 2024-12-31:
I can confirm the same issues. Firefox on linux doesnt work, both in flatpak and rpm. Chromium on linux works in both cases. Will test on macos and update.

patvdleer, 2025-01-06:
I fixed it for me, I run the non snap version of Firefox and came across https://askubuntu.com/a/1520568/21245

I've added the following to `/etc/apparmor.d/usr.bin.firefox`
```
  # FIDO
  /sys/class/hidraw/ r,
  /dev/hidraw* rw,

  # Yubikey NEO also needs this:
  /sys/devices/**/hidraw/hidraw*/uevent r,
```

and reloaded apparmor
```bash
sudo service apparmor reload
```

closed 2025-01-15 by polhenarejos