polhenarejos/pico-fido #67: Waveshare RP2040 One fails to register in system

iamthen0ise, 2024-10-06:
Hi!

Recently flashed pico-fido 5.12 patches as yubikey 4/5 onto waveshare dongle.
Succesfully tested onto webauthn.io and Yubi's test page, and  registered as 2fa device for 1password.
But i cant login into 1password anymore after flashing. Though test pages working good.

Under the hood i have noticed some weird behaviour.

sudo dmesg -Hw:
```
[  +0.000867] usb 3-5: New USB device found, idVendor=1050, idProduct=0407, bcdDevice= 6.00
[  +0.000001] usb 3-5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  +0.000001] usb 3-5: Product: Pico Key
[  +0.000001] usb 3-5: Manufacturer: Pol Henarejos
[  +0.000000] usb 3-5: SerialNumber: DE6350846737782D
[  +0.006627] hid-generic 0003:1050:0407.0058: hiddev96,hidraw1: USB HID v1.11 Device [Pol Henarejos Pico Key] on usb-0000:00:14.0-5/input0
[  +0.002443] hid (null): report_id 0 is invalid
[  +0.000081] hid-generic 0003:1050:0407.0059: report_id 0 is invalid
[  +0.000001] hid-generic 0003:1050:0407.0059: item 0 1 1 8 parsing failed
[  +0.000004] hid-generic 0003:1050:0407.0059: probe with driver hid-generic failed with error -22
```

this lead to device disconnection for pcscd:

```

04485487 ../PCSC/src/hotplug_libudev.c:646:HPEstablishUSBNotifications() USB Device add
00000072 ../PCSC/src/hotplug_libudev.c:299:get_driver() Looking for a driver for VID: 0x1050, PID: 0x0407, path: /dev/bus/usb/003/055
00000006 ../PCSC/src/hotplug_libudev.c:420:HPAddDevice() Adding USB device: Yubico YubiKey OTP+FIDO+CCID
00000034 ../PCSC/src/readerfactory.c:1103:RFInitializeReader() Attempting startup of Yubico YubiKey OTP+FIDO+CCID [Pico Key HID Interface] (DE6350846737782D) 00 00 using /usr/lib/pcsc/drivers/ifd-ccid.bundle/Contents/Linux/libccid.so
00000194 ../PCSC/src/readerfactory.c:977:RFBindFunctions() Loading IFD Handler 3.0
00000016 ifdhandler.c:2088:init_driver() Driver version: 1.5.5
00000473 ifdhandler.c:2110:init_driver() LogLevel: 0x0003
00000003 ifdhandler.c:2121:init_driver() DriverOptions: 0x0000
00000223 ifdhandler.c:111:CreateChannelByNameOrChannel() Lun: 0, device: usb:1050/0407:libudev:0:/dev/bus/usb/003/055
00000009 ccid_usb.c:331:OpenUSBByName() Using: /usr/lib/pcsc/drivers/ifd-ccid.bundle/Contents/Info.plist
00000485 ccid_usb.c:349:OpenUSBByName() ifdManufacturerString: Ludovic Rousseau (ludovic.rousseau@free.fr)
00000006 ccid_usb.c:350:OpenUSBByName() ifdProductString: Generic CCID driver
00000002 ccid_usb.c:351:OpenUSBByName() Copyright: This driver is protected by terms of the GNU Lesser General Public License version 2.1, or (at your option) any later version.
00002200 ccid_usb.c:724:OpenUSBByName() Found interface 2 but expecting 0
00000003 ccid_usb.c:726:OpenUSBByName() Wrong interface for USB device 3/55. Checking next one.
00000167 ccid_usb.c:208:close_libusb_if_needed() libusb_exit
00000034 ccid_usb.c:892:OpenUSBByName() Device not found?
00000004 ifdhandler.c:156:CreateChannelByNameOrChannel() failed
00000003 ../PCSC/src/readerfactory.c:1144:RFInitializeReader() Open Port 0x200000 Failed (usb:1050/0407:libudev:0:/dev/bus/usb/003/055)
00000003 ../PCSC/src/readerfactory.c:371:RFAddReader() Yubico YubiKey OTP+FIDO+CCID [Pico Key HID Interface] (DE6350846737782D) init failed.
00000003 ../PCSC/src/readerfactory.c:627:RFRemoveReader() UnrefReader() count was: 1
00000002 ../PCSC/src/readerfactory.c:1157:RFUnInitializeReader() Attempting shutdown of Yubico YubiKey OTP+FIDO+CCID [Pico Key HID Interface] (DE6350846737782D) 00 00.
00000002 ../PCSC/src/readerfactory.c:1014:RFUnloadReader() Unloading reader driver.
00000142 ../PCSC/src/hotplug_libudev.c:646:HPEstablishUSBNotifications() USB Device add
00000055 ../PCSC/src/hotplug_libudev.c:299:get_driver() Looking for a driver for VID: 0x1050, PID: 0x0407, path: /dev/bus/usb/003/055
00000004 ../PCSC/src/hotplug_libudev.c:420:HPAddDevice() Adding USB device: Yubico YubiKey OTP+FIDO+CCID
00000022 ../PCSC/src/readerfactory.c:1103:RFInitializeReader() Attempting startup of Yubico YubiKey OTP+FIDO+CCID [Pico Key HID Keyboard Interface] (DE6350846737782D) 00 00 using /usr/lib/pcsc/drivers/ifd-ccid.bundle/Contents/Linux/libccid.so
00000121 ../PCSC/src/readerfactory.c:977:RFBindFunctions() Loading IFD Handler 3.0
00000010 ifdhandler.c:2088:init_driver() Driver version: 1.5.5
00000370 ifdhandler.c:2110:init_driver() LogLevel: 0x0003
00000002 ifdhandler.c:2121:init_driver() DriverOptions: 0x0000
00000162 ifdhandler.c:111:CreateChannelByNameOrChannel() Lun: 0, device: usb:1050/0407:libudev:1:/dev/bus/usb/003/055
00000004 ccid_usb.c:331:OpenUSBByName() Using: /usr/lib/pcsc/drivers/ifd-ccid.bundle/Contents/Info.plist
00000372 ccid_usb.c:349:OpenUSBByName() ifdManufacturerString: Ludovic Rousseau (ludovic.rousseau@free.fr)
00000003 ccid_usb.c:350:OpenUSBByName() ifdProductString: Generic CCID driver
00000001 ccid_usb.c:351:OpenUSBByName() Copyright: This driver is protected by terms of the GNU Lesser General Public License version 2.1, or (at your option) any later version.
00002038 ccid_usb.c:724:OpenUSBByName() Found interface 2 but expecting 1
00000003 ccid_usb.c:726:OpenUSBByName() Wrong interface for USB device 3/55. Checking next one.
00000175 ccid_usb.c:208:close_libusb_if_needed() libusb_exit
00000029 ccid_usb.c:892:OpenUSBByName() Device not found?
00000004 ifdhandler.c:156:CreateChannelByNameOrChannel() failed
00000003 ../PCSC/src/readerfactory.c:1144:RFInitializeReader() Open Port 0x200000 Failed (usb:1050/0407:libudev:1:/dev/bus/usb/003/055)
00000003 ../PCSC/src/readerfactory.c:371:RFAddReader() Yubico YubiKey OTP+FIDO+CCID [Pico Key HID Keyboard Interface] (DE6350846737782D) init failed.
00000003 ../PCSC/src/readerfactory.c:627:RFRemoveReader() UnrefReader() count was: 1
[... 24 lines cut ...]
00000003 Card ATR: 3B FD 13 00 00 81 31 FE 15 80 73 C0 21 C0 57 59 75 62 69 4B 65 79 40
```

sudo usbhid dump says:
```
> sudo usbhid-dump -m 1050:0407

003:055:001:DESCRIPTOR         1728203024.597067
 05 01 09 06 A1 01 85 00 05 07 19 E0 29 E7 15 00
 25 01 95 08 75 01 81 02 95 01 75 08 81 01 05 08
 19 01 29 05 95 05 75 01 91 02 95 01 75 03 91 01
 05 07 19 00 2A FF 00 15 00 26 FF 00 95 06 75 08
 81 00 C0

003:055:000:DESCRIPTOR         1728203024.598741
 06 D0 F1 09 01 A1 01 09 20 15 00 26 FF 00 75 08
 95 40 81 02 09 21 15 00 26 FF 00 75 08 95 40 91
 02 C0
```

so i cant event figure out, is it a waveshare problem, or my hid device cant provide valid information to system (Linux arch 6.11.1-arch1-1). Have a little hope to make my mfa device to work. 

I would be appreciated with any help. Have a usb-ttl converter as well, maybe gpio could help to reinit device, but afaik it is not possible to emulate hid via serial.

Thank you!

polhenarejos, 2024-10-06:
There was a bug with the descriptor. Can you try with development nightly build?

iamthen0ise, 2024-10-06:
Works like magic now, thank you! So sad my 1password account gone to freeze, hope for support from them.

closed 2024-10-07 by iamthen0ise