polhenarejos/pico-fido #69: Hardware recommendations

patvdleer, 2024-10-29:
Any thoughts on hardware?

I was looking at the Waveshare RP2040-One as a usb stick, anything with a male USB A or C would be nice. Any thoughts or recommendations?

https://www.waveshare.com/wiki/RP2040-One

For NL/EU, €5,75, 
https://www.tinytronics.nl/en/development-boards/microcontroller-boards/others/waveshare-rp2040-one-4mb-flash

air-eat, 2024-11-10:
i'm using an rp2040-one as a fido2 key currently; they're fairly nice rp2040 based boards that cost only a euro or two more than a regular pico

i have two main problems with it: the pcb usb connector can sometimes not make contact with usb ports, and the reset (restart) button is very easy to press by mistake instead of the boot button - apart from that, i personally think they're a good choice due to their cost and size

niutech, 2024-12-20:
There are some dirt-cheap USB stick boards on AliExpress:
- [LC Raspberry Pi RP2040 USB](https://www.aliexpress.com/item/1005007557339976.html) for $3.00
- [RP2350-One](https://www.aliexpress.com/item/1005008258590162.html) for $6.16

polhenarejos, 2024-12-20:
Go for RP2350. It has hardware security specially to protect the keys.

patvdleer, 2024-12-22:
> There are some dirt-cheap USB stick boards on AliExpress:
> 
>     * [LC Raspberry Pi RP2040 USB](https://www.aliexpress.com/item/1005007557339976.html) for $3.00
> 
>     * [RP2350-One](https://www.aliexpress.com/item/1005008258590162.html) for $6.16

> Go for RP2350. It has hardware security specially to protect the keys.

So I bought 3 of the RP2350 @niutech linked, 3 for € 5.89 each, € 4.02 shipping (to NL), so €21.69 in total.

niutech, 2024-12-22:
@patvdleer You could get a better price at $4.50 for 3+ pcs on [Alibaba](https://www.alibaba.com/product-detail/RP2350-One-Waveshare-Onboard-PCB-Type_1601324274925.html).

polhenarejos, 2025-01-15:
No activity. Reopen if needed.

closed 2025-01-15 by polhenarejos

mmaret-psee, 2025-01-20:
Would it make sens to have this in the README.md ? Like a recommendation for "we know its working correctly on this device" ?

niutech, 2025-01-20:
I've bought Waveshare RP2350-One, flashed Pico Fido (for Raspberry Pi Pico 2) and it works fine, except the WS2812 RGB LED (it has to be supported in Pico SDK).

[image]

[image]