polhenarejos/pico-fido #70: Online vid/pid firmware editing does not work!

metabo7000, 2024-11-12:
Online vid/pid firmware editing does not work!
windows 10..11 online vid/pid firmware editing does not work!
nothing happens after webusb!
Why did you kill the old online vid/pid tool the good work?!
how to use this?
video
[https://drive.google.com/file/d/1a1OicQoEnScrFCfF44a_sXuE_YAq7Qqc/view?usp=drive_link](url)

Supermath101, 2024-11-29:
Try following [these instructions](https://github.com/polhenarejos/pico-fido/issues/73#issue-2685086542) to use an alternative method, which is a Python script found in this GitHub repo.

metabo7000, 2024-11-29:
> Try following [these instructions](https://github.com/polhenarejos/pico-fido/issues/73#issue-2685086542) to use an alternative method, which is a Python script found in this GitHub repo.
Then you could show me the instructions!

metabo7000, 2024-11-29:
> +1 I'm using ESP32-S3 and facing the same issue with Commission via USB and Commission via WebAuthn.
> 
> Appreciate any help please.

Above v5.8 everything is broken and useless!

Supermath101, 2024-11-29:
> > Try following [these instructions](https://github.com/polhenarejos/pico-fido/issues/73#issue-2685086542) to use an alternative method, which is a Python script found in this GitHub repo.
>
> Then you could show me the instructions!

Run these commands:
1. `fido2-token -L`
2. `fido2-token -S <device>` for which `<device>` is found in the previous step.
3. `python3 -m pip install fido2`
4. `git clone` this repo.
5. `cd pico-fido`
6. `python3 tools/pico-fido-tool.py --help` to see the options for configuration, and what command is necessary to run to configure them.

metabo7000, 2024-11-29:
> 6\. pico-fido-tool.py

write down the correct command as it should be! pico-fido-tool.py test1.uf2?

Supermath101, 2024-11-29:
> > 6. pico-fido-tool.py
> 
> write down the correct command as it should be! pico-fido-tool.py test1.uf2?

`python3 tools/pico-fido-tool.py -p <PIN> phy vidpid <VID>:<PID>`, replacing `<PIN>` with your pin that you must've set in step two, and `<VID>` with the four hexadecimal digit VID, and `<PID>` with the four hexadecimal digit PID.

metabo7000, 2024-11-29:
> > > 6. pico-fido-tool.py
> > 
> > 
> > write down the correct command as it should be! pico-fido-tool.py test1.uf2?
> 
> `python3 tools/pico-fido-tool.py -p <PIN> phy vidpid <VID>:<PID>`, replacing `<PIN>` with your FIDO2 PIN that you should've set in step two, and `<VID>` with the four hexadecimal digit USB VID, and `<PID>` with the four hexadecimal digit USB PID.

pico-fido-tool.py -p 1234 phy vidpid 1050:0407
[
[image]
](url)

Supermath101, 2024-11-29:
> pico-fido-tool.py -p 1234 phy vidpid 1050:0407 [ [...]

I've noticed that certain changes, such as the VID/PID combo, only go into effect after unplugging and replugging-in the Pico. I forget whether that was also the situation for the FIDO2 PIN change, but it might be. Beyond that, I've only tested this myself on Linux, so that could be the issue.

PS I'm not the one who developed any of this. I just reverse engineered it.

metabo7000, 2024-11-29:
> > pico-fido-tool.py -p 1234 phy vidpid 1050:0407[ ![kép](https://private-user-images.githubusercontent.com/171061281/391195300-e1eee79b-1051-4752-8e44-e44effa253cc.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzI5MjAyMTAsIm5iZiI6MTczMjkxOTkxMCwicGF0aCI6Ii8xNzEwNjEyODEvMzkxMTk1MzAwLWUxZWVlNzliLTEwNTEtNDc1Mi04ZTQ0LWU0NGVmZmEyNTNjYy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTI5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTEyOVQyMjM4MzBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1hODM1OThhM2U4ZTE0MDcxOWE0YmNkMDhkM2Y5MDgxZjBkNjcwNmUxYWM4MzI4NTRiMjYxOTU4ZjgxNTYzMjZmJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.7-A2ENT7iMbBd_MGZru4PXOrUIEiwFcKXcrOQkdDXis) ](url)
> 
> Észrevettem, hogy bizonyos változtatások, mint például a VID/PID kombó, csak a Pico kihúzása és újracsatlakoztatása után lépnek életbe. Elfelejtettem, hogy ez volt-e a helyzet a FIDO2 PIN cseréjénél is, de lehet, hogy így van. Ezen túlmenően ezt csak Linuxon teszteltem, szóval ez lehet a probléma.
> 
> PS Nem én vagyok az, aki ezt kifejlesztette. Csak visszafejtettem.

Thank you for your help! This project is only getting worse v6 is already catastrophic look back sea of ​​problems!

closed 2025-01-15 by polhenarejos