polhenarejos/pico-fido #72: Can't create Google passkey in pico_fido_pico-5.12.uf2 & pico_fido_pico-6.0.uf2

GremlinStyle, 2024-11-21:
Hello,
And first i would like to thank you for this awesome project

I noticed an error/bug (or my fault?) which doesn't allow me to create a google passkey in version 5.12 and 6.0 (not nightly) but in 5.8 it is working
I tested it on waterfox and chrome on my windows 10 pc
<code>OS Name:                   Microsoft Windows 10 Home
OS Version:                10.0.19045 N/A Build 19045
OS Manufacturer:           Microsoft Corporation
OS Configuration:          Standalone Workstation
OS Build Type:             Multiprocessor Free</code>

After clicking "create passkey" and entering my security pin
I was stuck for some minutes at this step
[image]
Before i finally get the message "Couldn't complete singup" or something similar 
(I couldn't catch a screenshot because it's so fast gone)

I hope this information is in someway helpful and wish you a nice evening/morning/day (where ever you are)
(this is my first time writing an issue so i hope it's at least somewhat usable)

polhenarejos, 2024-11-22:
Does it work with webauthn.io ?

GremlinStyle, 2024-11-22:
Yes i did test it on the 6.0 & 5.12 with webauthn.io or https://www.token2.com/tools/fido2-demo and it always worked flawless

polhenarejos, 2024-11-22:
I've tried right now with Chrome in macOS and worked. Perhaps is the middle layer that Windows use. 

Can you try with macOS or Linux?

GremlinStyle, 2024-11-22:
This is a little embarrassing but it seems i can't use on a freshly new installed ubuntu desktop (on my laptop).

First i tried without installing anything but then after some research installed these
pscscd pscsc-tools sssd libpam-sss opensc-pkcs11

And i still can't use it (even after repeatedly reflashing with 6.0 and 5.8 version)
I tried it on firefox and chromium before digging deeper and i found out that

in most cases the pico doesn't even get detected
at least not with pcsc_scan and the browsers

Only from pamu2fcg did i get some success which is another error message
<code>$ pamu2fcfg
No U2F device available, please insert one now, you have  6 seconds
Device found!
Enter PIN for /dev/hidraw0:
error: fido_dev_make_cred (39) FIDO_ERR_OPERATION_DENIED</code>

Or in case i press the button after entering the key without waiting for any message
<code>$ pamu2fcfg
Enter PIN for /dev/hidraw0:
error: fido_cred_verify (-7) FIDO_ERR_INVALID_ARGUMENT</code>

In case i forgot to install a driver or maybe to configure a service let me know?

And on a site note is it normal for the pico to be detected as smartcard by windows?
I thought it is more like a fido key( yubikey and so ) i noticed it while trying to detect it with libfido2 for 2 hours before seeing that windows calls it a smartcard

polhenarejos, 2024-11-22:
For being recognized by OpenSC, you must commission it with known VID & PID. More info at https://www.picokeys.com/pico-commissioner/
Note that using known VID & PID is only necessary for CCID interface and 3rd party tools like Yubikey Manager or similar. FIDO should work with any VID & PID. It has not been tested with PAM.

Pico Fido has FIDO, Keyboard, CCID and WebCCID interfaces, so yes, it is normal.

When I said "try with macOS or Linux" I referred to create Google passkey using Chrome in macOS or Linux.

GremlinStyle, 2024-11-23:
Thanks for the explanation
And maybe i said it poorly but i meant i can't use the pico-fido at all in firefox/chrome on ubuntu it just doesn't get detected.
I tried first with webauthn.io but and get the message in firefox to press the button, nothing happens if i do and in chromium to insert the key.

So i tried to troubleshoot it with pamu2fcfg.
I don't know if for linux i forgot to install some packages or similar.

GremlinStyle, 2024-11-23:
Ok now also tested on linux mint with chromium.
with libu2f-udev pcscd installed and that 70-u2f.roles file

Again tested with version 6.0 and 5.8

**6.0:**
Webauthn.io works fine but in chrome but for the google passkey i don't get past the "Press the button on your key" part
It just doesn't react to the button press
similar to the windows case.

**5.8:**
It works with google just fine

polhenarejos, 2024-11-23:
But do you press the BOOT button to confirm? Not the reset one.

GremlinStyle, 2024-11-24:
Yes the button with the text "BOOTSEL" above it also the only button onboard
[image]

polhenarejos, 2024-11-24:
Are you using Pico board? The same as the pic.

GremlinStyle, 2024-11-24:
Exactly, it is the same.

polhenarejos, 2024-11-25:
Seems a problem of timeout. 
In webauthn.io (and probably others) the process is this:
- You click on Register. Board will enter in "waiting for button" state.
- Press BOOTSEL button in less than 10 seconds; otherwise will fail with timeout.
- A PIN windows will appear
- Introduce your PIN and continue. Board will enter in "waiting for button" state.
- Press BOOTSEL button again in less than 10 seconds; otherwise will fail with timeout.
- The process will conclude successfully.

Can you confirm you press BOOTsel button **twice** in **less than 10 seconds** after click on Register/Authenticate **and** PIN input?

GremlinStyle, 2024-11-25:
Strange,
I tested it on windows 11 and it worked just fine
But on windows 10 it's getting stuck after entering the pin

So i don't even get to the "waiting for button" state.

This is what i do.

Expected happenings
[image]
[image]
[image]
[image]

Here it's stuck for a minute or two
[image]

And error messages following
[image]
[image]

windskyxb, 2024-11-26:
Same issue
[image]
[image]

polhenarejos, 2024-11-26:
But do you press twice the button? One after click on "Register" and one after introducing the PIN.

windskyxb, 2024-11-26:
> But do you press twice the button? One after click on "Register" and one after introducing the PIN.

After entering the PIN for the security key, the system will get stuck in `getting things ready` for a while and report an error: `Could't complete sign-in. Try again and ensure you do the follow-up action.` Prior to this, I had already used Pico Commissioner to set VENDOR to Yubikey 4/5. During this process, the system did not require me to press a button

GremlinStyle, 2024-11-26:
I tested if i can login in google after creating the key on windows 11.
And it worked.

But if i try to create the key it fails.

Also it (creating the key) doesn't work in linux mint with firefox or chromium

GremlinStyle, 2024-11-27:
Update:

I tested it today it on a <a href=https://www.raspberrypi.com/products/raspberry-pi-pico-2/>PICO 2</a> with the firmware <a href=https://github.com/polhenarejos/pico-fido/releases/download/v6.0/pico_fido_pico2-6.0.uf2>6.0</a>

On Windows 10:
Firefox and Chrome (same responses):
For <a href=https://webauthn.io/>webauthn.io</a> i get stuck at the same <code>"getting things ready"</code> error
but for <a href=https://www.token2.com/tools/fido2-demo>fido2-demo</a> it's working
google doesn't even find it, error message <code>"Are you there? Use your security key to sign in.</code>

Linux Mint:
Chrome can create a key but only on <a href=https://webauthn.io/>webauthn.io</a> but nothing else
Registered key can be used to login on any OS or browser.
Firefox doesn't work at all, like windows firefox

GremlinStyle, 2024-11-28:
Hello,
It's been a while today i gathered some useful troubleshooting infos from the windows even viewer
Here are the logs as txt, xml and windows event viewer format [pico_fido.zip](https://github.com/user-attachments/files/17952031/pico_fido.zip)

Also i used chrome to gather additional data
[image]

I hope these are use full

polhenarejos, 2024-11-28:
1. Install the development nightly build, as it includes a new capability to tune the button timeout.
2. Go to the Commissioner and set the Presence Button Timeout to 0 to disable it.
3. Test again to see whether it is a problem with the button or it is another timeout.

GremlinStyle, 2024-11-29:
Anyway even after flashing the <a href=https://github.com/polhenarejos/pico-fido/releases/download/nightly-development/pico_fido_pico-6.0.uf2>file</a> onto the pico and afterwards using the commission tool to set the timeout to 0

In linux it works with webauthn.io but google still has the problem where it loads without any end in sight.
Led is still blinking.

In windows 10 it sometimes works on webauthn.io? but only once then i could not replicate it.
It still is the same problem where the pico seems to crash (led not blinking, either stuck at glowing or being not on) sometimes.
And that "getting things ready" error is ever present.

i added here the chrome logs from windows 10 chrome
chrome://device-log/?refresh=5
```
FIDOEvent[20:58:14] UI step: kClosed
 
FIDODebug[20:58:14] Advertisements stopped
 
FIDODebug[20:58:14] Stopping 0 caBLE advertisements
 
FIDODebug[20:58:14] WebAuthNAuthenticatorMakeCredential()=0x800704C7 (NotAllowedError)
 
FIDODebug[20:55:21] WebAuthNAuthenticatorMakeCredential(rp={1, "webauthn.io", "webauthn.io"}, user={1, 3654314B79305266534F793179737339346E7832686A5361306E64346437515F376C664F416E5161467273, "test", "test"}, cose_credential_parameters={2, &[{1, "public-key", -7}, {1, "public-key", -257}]}, client_data={1, "{\"type\":\"webauthn.create\",\"challenge\":\"Ok4fC7GKEQirJN14yAGI0EDaCYaWD6ZcIOT-pvTfHMT-6rh3VCSqDFl9AV7N-7BroCUfxrDDbI_y2aFRApTF7w\",\"origin\":\"https://webauthn.io\",\"crossOrigin\":false}", "SHA-256"}, options={7, 300000, {0, &[]}, {0, &[]}, 2, 0, 2, 1, 0, 0000008E095FF2E0, &{20, &[&{1, 1889BAAF3A93708863CC4487EE3C1B46, "public-key", 48}, &{1, 2412669E3D28C8B120202E13F963F34417801B0A, "public-key", 16}, &{1, 255E698DA8795859CCC4D8F83869B95F0E526BE4, "public-key", 48}, &{1, 27C824054F36BBF0F5B5C7F6C5D91D81, "public-key", 48}, &{1, 51C91DFB590B770B10F1838D3B3B5EDE, "public-key", 48}, &{1, 7BD9DFF7A5BFDC681131A7ED86945C171263D2823CFDB5850EC368B55EAD45751D3E74A6EA9213C99C2F74B22492B320CF40262A94C1A950A0397F29250B60841EF01D000000, "public-key", 55}, &{1, 83BB2B166E6A1F37B218E1A4BA5E7900D39F02D0AE7EC397C47ED306CFDF649C, "public-key", 16}, &{1, 8E86DA1548CF4A5EB34747B60FF30D80, "public-key", 16}, &{1, 9400E086967BFC3BFE99768219054917, "public-key", 48}, &{1, 970A68F4A3CDD76457696B15F170876459E2504321342C0C5A0FFBA0E3CAD15A0B2F74A6EA9213C99C2F74B22492B320CF40262A94C1A950A0397F29250B60841EF01A000000, "public-key", 55}, &{1, 9BD04F94DB162834D75A9FBE370BFFDC3BA53FE6A9FC9BE9034AFD178011F8F6, "public-key", 16}, &{1, AA3368683832CEDEB560D7B1403400F98F54C608605ACC27424099A02142AFEF2153B5B67457C3A6ED3D40D40B98E1BE, "public-key", 3}, &{1, CC44AC17DEC945BED438CE656DE0EE96, "public-key", 3}, &{1, DD5B79D62A70487C89DDEEFADFB9D72D, "public-key", 16}, &{1, E523D1C3DD2B661064CB1967E66D60777EDB80D70A823A24E49992CF09F8258DBE1474A6EA9213C99C2F74B22492B320CF40262A94C1A950A0397F29250B60841EF01A000000, "public-key", 55}, &{1, F1D0020124A3935A14ED944ADF38D66629C4C6E941FFD02B476F79A7A38D3447F2F06AC831DEEFB1733D3AD08374F9538D5092B68112E98FC6D5408E648A41674BFFC760DAB08C1F9A4732F52C9266ED821581ED8F0EF1613AB3BF50F5A851DCA8FFD343D25FDF3927886751C0221771485D767CECE65B8826DE2D9684A0BFE47BA8D5481AD5B9CA6051AEAE9CDFC58BCBEE6EFC70E6AE9FD73259623C2A2849259E, "public-key", 1}, &{1, F1D002013206C2FAD750EA4E5F59B8DF708DC230C09896654ED5791880FC03C8D10CA3E130D543C3BCCE87D8BB98B9BD50ACEC85392C77819371107D40F7A901BD1CD6A0477D2A64573D91C699C0252521949B404F2AC85360BC037836F74A98FBA17072A661900302E0D2D488C08A3DD0FB60CC041B0E423F8DE2BF613C7AF54E3A2DAA14BE645CCE73D88E654B97D7FE0B74CF5D3410D41321B237434C08B9391A, "public-key", 1}, &{1, F1D002015B3563667E9B9B1D645B98E3480701AE55AC9E9B2E1F37CEDA36E058A54B28205B6168FA01CE2F4A4074D1E64BC55B725162E4200DBB753414CA7ACDF594F776ABCE9D79E3877E3416F819185355A964F56813CEFB3AD440260B82F81523E19D626B35B6456CEE4F0A92F098952FA360FC9D85DB3D95BE4D64BC642D79793FA8FE996329DA622FD05CECA35DA6839F5BB6CF8451BB910790B32C777F158D3C13A16BF8487CDCBFC176D605ABEE4B8231ACEA5278D9, "public-key", 55}, &{1, F4364E04D82E4D7CA5A15933EFA1442A, "public-key", 16}, &{1, FA8B20609C0A014A5D9E258A2EF583A7A3CF88C4, "public-key", 16}]}, 0})
 
FIDOEvent[20:55:21] UI step: kNotStarted
 
FIDODebug[20:55:21] No BLE adapter present
 
FIDODebug[20:55:21] Bluetooth status: Off
 
FIDOEvent[20:55:21] Starting MakeCredential flow: { "attestation": "none", "authenticatorSelection": { "authenticatorAttachment": "cross-platform", "residentKey": "preferred", "userVerification": "preferred" }, "challenge": "Ok4fC7GKEQirJN14yAGI0EDaCYaWD6ZcIOT-pvTfHMT-6rh3VCSqDFl9AV7N-7BroCUfxrDDbI_y2aFRApTF7w", "excludeCredentials": [ { "id": "GIm6rzqTcIhjzESH7jwbRg", "transports": [ "hybrid", "internal" ], "type": "public-key" }, { "id": "JBJmnj0oyLEgIC4T-WPzRBeAGwo", "transports": [ "internal" ], "type": "public-key" }, { "id": "JV5pjah5WFnMxNj4OGm5Xw5Sa-Q", "transports": [ "hybrid", "internal" ], "type": "public-key" }, { "id": "J8gkBU82u_D1tcf2xdkdgQ", "transports": [ "hybrid", "internal" ], "type": "public-key" }, { "id": "Uckd-1kLdwsQ8YONOzte3g", "transports": [ "hybrid", "internal" ], "type": "public-key" }, { "id": "e9nf96W_3GgRMafthpRcFxJj0oI8_bWFDsNotV6tRXUdPnSm6pITyZwvdLIkkrMgz0AmKpTBqVCgOX8pJQtghB7wHQAAAA", "transports": [ "usb", "nfc", "ble", "hybrid", "internal" ], "type": "public-key" }, { "id": "g7srFm5qHzeyGOGkul55ANOfAtCufsOXxH7TBs_fZJw", "transports": [ "internal" ], "type": "public-key" }, { "id": "jobaFUjPSl6zR0e2D_MNgA", "transports": [ "internal" ], "type": "public-key" }, { "id": "lADghpZ7_Dv-mXaCGQVJFw", "t [... cut ...]
USBEvent[20:55:01] USB device function updated: guid=43c1a89f-7166-45d2-a329-d07daed49109, interface_number=2, path="\\?\usb#vid_20a0&pid_42b1&mi_02#6&62bb226&0&0002#{50dd5230-ba8a-11d1-bf5d-0000f805f530}", driver="WUDFRd"
 
USBUser[20:55:01] USB device added: path=\\?\usb#vid_20a0&pid_42b1#e660382823638f35#{a5dcbf10-6530-11d2-901f-00c04fb951ed} vendor=8352 "Pol Henarejos", product=17073 "Pico Key", serial="E660382823638F35", driver="usbccgp", guid=43c1a89f-7166-45d2-a329-d07daed49109
``` 

Also checked if it's not the pico which may be damaged but with the 5.8 version it's working just fine.

Tested on two windows 10 pc
And a linux mint

PS:
Thanks for taking your time to work this issue

polhenarejos, 2024-11-29:
What happens if you run the tests?

`pytest tests`

GremlinStyle, 2024-11-30:
Done
I should have started with this right?
Anyway here are the logs for 6.0 with and without the comission where the timout is set to 0.
[picofido_6.0_night_dev_with_comission.log](https://github.com/user-attachments/files/17965553/picofido_6.0_night_dev_with_comission.log)
[picofido_6.0_night_dev_without_comission.log](https://github.com/user-attachments/files/17965554/picofido_6.0_night_dev_without_comission.log)

Enviroment is windows and scripts are run with admin privs.
Otherwise pytest gives me an Access denied error.

polhenarejos, 2024-12-04:
Try using the nightly build and disable the Power cycle after reboot option in the Commissioner.

GremlinStyle, 2024-12-07:
Hi
Sorry for the delay, 
It is still not working as intended (as in getting stuck at "Getting things ready" even on webauthn.io)

Here the logs and i wasn't sure which nightly so i did both:
[PicoFido_nightly-dev_Commision-PowerCycle_7-12-2024_windows10.log](https://github.com/user-attachments/files/18048527/PicoFido_nightly-dev_Commision-PowerCycle_7-12-2024_windows10.log)
[PicoFido_nightly-stable_Commision-PowerCycle_7-12-2024_windows10.log](https://github.com/user-attachments/files/18048528/PicoFido_nightly-stable_Commision-PowerCycle_7-12-2024_windows10.log)

Thanks

79812b, 2024-12-10:
hey i am facing the same problem 
Win10

when using on discord or webauth.io it works

but when trying to use it on binance or google it get stuck in "getting things ready phase" 
if anyone have any solution let me know

polhenarejos, 2024-12-11:
But this is the expected behavior. If a credential is already created and relayed by the RP, then it fails creating a new one (because it has been already created). It is described here:
https://fidoalliance.org/specs/fido-v2.1-ps-20210615/fido-client-to-authenticator-protocol-v2.1-ps-errata-20220621.html#op-makecred-step-if-excludeList

I already created passkeys in google without problems.

jcodeth, 2024-12-11:
Sorry for the poor explanation.
I wanted to explain why it works with webauthn.io.
I think the problem may be caused by omission of initialization of the structure.
https://github.com/polhenarejos/pico-fido/blob/7a59b518494d46e01dd768e5d9af4a9caaf3bd74/src/fido/credential.c#L146-L160

/src/fido/cbor_make_credential.c
```C
    Credential ecred = {0};    // Add initialization
    if (credential_load(excludeList[e].id.data, excludeList[e].id.len, rp_id_hash,
                             &ecred) == 0 &&
             (ecred.extensions.credProtect != CRED_PROT_UV_REQUIRED ||
              (flags & FIDO2_AUT_FLAG_UV))) {
                 credential_free(&ecred);
             CBOR_ERROR(CTAP2_ERR_CREDENTIAL_EXCLUDED);
    }
    credential_free(&ecred);
```

polhenarejos, 2024-12-11:
I tried to create a new passkey in Google without problems. Here's the flow (it's in catalan but easily recognized):

1. Two options: "Create a passkey" or "Use a security key". Click on "Use a security key", the good one.
[image]
2. Configure this key to be used by Google. OK or NOK?
[image]
3. Google will get the vendor name and model. Are you sure?
[image]
4. Introduce the PIN
[image]
5. Touch the BOOTSEL button.
[image]
And voilà. After touching the key is registered.

polhenarejos, 2024-12-11:
You're right @jcodeth . If it is not properly initialized, it might crash on credential_free() afterwards. I'll push a fix.

samsell, 2024-12-11:
I think I have the same issue, arch linux using chrome 131. Firefox doesn't work at all, but that looks expected from some of the other issues. Works perfectly fine with github and webauthn.io, but creating a passkey for a google account fails at the same point:
[image]

Pretty sure it is properly initialised, through the web commissioner

polhenarejos, 2024-12-11:
I found the bug and it is related with the timeout/keepalive procedure. By specified in the FIDO documents, the authenticator must send keepalive commands regularly, to inform the client the command is still being processed. But taking the old U2F specifications, no keepalive was specified. So, it is in the client side the procedure to check the timeout.

Depending on the platform, google may decide to use CTAP2 (FIDO) protocol or CTAP1 (U2F). If it chooses CTAP2 (my above examples), it works normally. But if it chooses CTAP1, then it does not expect any keepalive command, regardless the specification. Once the keepalive is sent, it throws an error about "unknown command" and closes the connection. 

I put a fix which mainly sends a keepalive command only if we are in a middle of CTAP2 transaction. Hopefully it will fix all these errors you have.

79812b, 2024-12-12:
oh thanks for the update this issue i was facing with
Google
Binance
mostly both having same issue hope this fixes binance one too

79812b, 2024-12-12:
@samsell how can i update my pico ? where can i get the new file?

samsell, 2024-12-12:
@79812b it's in the releases section, click the nightly development build and look for the pico-fido .uf2 file. Or the link is here if you have the standard pico: https://github.com/polhenarejos/pico-fido/releases/download/nightly-development/pico_fido_pico-6.0.uf2
[image]

Filz0r, 2024-12-12:
> @79812b it's in the releases section, click the nightly development build and look for the pico-fido .uf2 file. Or the link is here if you have the standard pico: [...]

Hello I was trying this project out on my pico on Firefox and Chromium on Linux and was having this exact issue, so I started to go trough this thread and found this message, tried it out and now it works, I'm now able to use my pico to authenticate for me, I was only able to register the device on Chromium though but that's already been mentioned elsewhere if I'm not mistaken.

Thanks a lot!

commit 022503fd (2024-12-11): In pure U2F mode, no keepalive is sent by authenticator. Instead, client sends commands to know the status. Fixes #72.
commit dba805dc (2024-12-11): Fix potential overflow due to bad initialization. Might fix #72.