polhenarejos/pico-fido #73: Commissioning via the `tools/pico-fido-tool.py` script doesn't work unless a PIN is set

Supermath101, 2024-11-23:
I'm using a Raspberry Pi 5 running Raspberry Pi OS 64-bit. For some reason, the [Pico Commissioner](https://www.picokeys.com/pico-commissioner/) doesn't work on Brave browser nor Firefox. So, I tried the alternative method of using the [`tools/pico-fido-tool.py`](https://github.com/polhenarejos/pico-fido/blob/v6.0/tools/pico-fido-tool.py) script. It requires a PIN as an argument. However, the PIN is unset by default. To set the PIN, I had to run `fido2-token -S /dev/hidrawX`, where `X` represents which USB device it is. Then, I was finally able to pass in the PIN to the `pico-fido-tool.py` script.

If you want to reproduce this, you'll first have to factory reset the Pico 2, which you can do by following these instructions: https://www.raspberrypi.com/documentation/microcontrollers/pico-series.html#resetting-flash-memory

polhenarejos, 2024-12-04:
This is intended. PIN is usually set during the first registration on any website through a browser, which is the primary purpose of any Fido key. Are you able to run it through the browser?

Supermath101, 2024-12-04:
> This is intended. PIN is usually set during the first registration on any website through a browser, which is the primary purpose of any Fido key. Are you able to run it through the browser?

If you're referring to the Pico Commissioner, then no. It shows a user interface, but it doesn't push the changes properly.

79812b, 2024-12-10:
for me pico commissioner works on brave

closed 2025-01-15 by polhenarejos