polhenarejos/pico-fido #75: Add support for working like 'FIPS YubiKeys'

kholia, 2024-11-26:
https://www.yubico.com/products/yubikey-fips/ are different from regular YubiKey(s).

Some environments only accept these 'FIPS YubiKey(s)'.

Thanks!

polhenarejos, 2024-11-26:
What do you mean by "adding support"? FIPS devices are more restrictive and limited, as they contain secure measures oriented to obtain a certification, which is not the scope of Pico Keys.

kholia, 2024-11-26:
Yes, Pico Keys may not gets FIPS certification but can the underlying `secure measures` be implemented to achieve unofficial compatibility / interoperability?

I don't know the inner working of these keys (yet) so please feel free to close this feature request.

Thank you!

jay0lee, 2024-12-01:
There'd be little point to unofficial FIPS support, it's either FIPS certified or it's not. Certification is a multi step process that involves 3p vendor review (read that as $$$).

closed 2024-12-18 by polhenarejos